karawin/Ka-Radio32 #2: Compile error bt_speaker

ArnostDvorak, 2017-10-20:
C:/msys32/home/user/Ka-Radio32/components/bt_speaker/bt_speaker.c: In function 'bt_av_hdl_stack_evt':
C:/msys32/home/user/Ka-Radio32/components/bt_speaker/bt_speaker.c:87:36: error: 'CONFIG_BT_NAME' undeclared (first use in this function)
         esp_bt_dev_set_device_name(CONFIG_BT_NAME);
                                    ^
C:/msys32/home/user/Ka-Radio32/components/bt_speaker/bt_speaker.c:87:36: note: each undeclared identifier is reported only once for each function it appears in

I used new chaintools and actual repository.

karawin, 2017-10-20:
KaRadio32 don't use BT fo now.
Remove it in menuconfig.
A bug in the idf is detected, so the web and telnet dont work anymore.
I patched it, a correction is in progress.
See https://github.com/espressif/esp-idf/issues/1141

karawin, 2017-10-30:
No answer. Closing

closed 2017-10-30 by karawin

ArnostDvorak, 2017-12-11:
Could you include the bluetooth music playback feature? Please.
Thaks

karawin, 2017-12-11:
I tried but unsuccessful, not enough memories.