karawin/Ka-Radio32 #4: Wiring

tomasf71, 2017-11-05:
Hi Jean-Pierre, I tried to transfer your text description of wiring to scheme. Could you check it. When I finish it, I can share with you to public in your repository. 
[image]

karawin, 2017-11-05:
I think that the led must be connected to gnd and not 3.3v. I like a very small flash on it.
Great job.
Not finished to check.

karawin, 2017-11-05:
A connector for lcd I2C?

tomasf71, 2017-11-05:
[image]

karawin, 2017-11-05:
Seems ok now.

tomasf71, 2017-11-05:
Thank you for your checking. Here you are. If anything to change, just let me know.
[scheme.pdf](https://github.com/karawin/Ka-Radio32/files/1444448/scheme.pdf)

closed 2017-11-05 by tomasf71