karawin/Ka-Radio32 #5: version 0.7 not update 0.7 R3

arlaor, 2017-11-19:
Hi @karawin 
I am trying to update the version to be able to listen in vs1053, but it does not update the new version.
I tried to upload the bin file again with the esp32 download tool, to update to the R3 version, but only the 0.7 version still appears.
version 0.7 does not allow me to play stations with AAC extension, I do not have sound in my speaker. I have used the pins suggested in the image, but I can not get it working.
The following shows it in the arduino IDE

##CLI.STOPPED# from unsupported mime type
##CLI.STOPPED# from OTA
Starting update ...
must receive:770384 bytes

Connection closed, all packets received
Total Write binary data length : 770384
Update firmware succeded. Restarting
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:2980
load:0x40078000,len:0
load:0x40078000,len:11400
entry 0x40078c38
I (324) main: starting app_main()
I (324) main: RAM left: 171968
E (324) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (354) main: Partition init done...
Log level is now ESP_LOG_ERROR
E (5464) vs1053: VS1053/VS1003 detected. Version: 0
device: 3ffc9e7c
volume: 128
E (5784) wifi: esp_wifi_disconnect 844 wifi not start

DNS: 192.168.1.254  
IP: 192.168.1.60

Init .................... Done
autostart: playing:1, currentstation:2
##CLI.NAMESET#: 2 80srock RR (rockradio.com) aac 128 kbps
##CLI.URLSET#: prem3.rockradio.com
##CLI.PATHSET#: /80srock_aac?a75b7607061c2551c187ccbe
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 128
##CLI.ICY0#: 80s Rock - ROCKRADIO.COM Premium
##CLI.ICY3#: http:\/\/www.rockradio.com
##CLI.ICY4#: Rock 80s Classic
##CLI.ICY5#: 128
##CLI.ICY7#:  ice-bitrate=128
##CLI.PLAYING#

##CLI.META#: Bruce Springsteen - My Hometown
E (17284) audio_player: failed to start decoder task
##CLI.STOPPED# from unsupported mime type
##CLI.NAMESET#: 1 80 HotMixRadio (80's) mp3 128 kbps
##CLI.URLSET#: streaming.hotmix-radio.net
##CLI.PATHSET#: /hotmixradio-80-128.mp3
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 128
##CLI.ICY0#: Hotmixradio 80
##CLI.ICY3#: http:\/\/www.hotmixradio.fr\/80
##CLI.ICY4#: 80s
##CLI.ICY5#: 128
##CLI.ICY7#:  ice-samplerate=44100;ice-bitrate=128;ice-channels=2
##CLI.PLAYING#

##CLI.META#: SCORPIONS - Still Loving You 
##CLI.STOPPED# from Player stop
##CLI.NAMESET#: 6 alternativerock RR (rockradio.com) aac 128 kbps
##CLI.URLSET#: prem3.rockradio.com
##CLI.PATHSET#: /alternativerock_aac?a75b7607061c2551c187ccbe
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 128
##CLI.ICY0#: Alternative Rock - ROCKRADIO.COM Premium
##CLI.ICY3#: http:\/\/www.rockradio.com
##CLI.ICY4#: Alternative Rock
##CLI.ICY5#: 128
##CLI.ICY7#:  ice-bitrate=128
##CLI.PLAYING#

##CLI.META#: Muse - Uprising
E (1196264) audio_player: failed to start decoder task
##CLI.STOPPED# from unsupported mime type

karawin, 2017-11-19:
Something is wrong  on the spi bus
E (5464) vs1053: VS1053/VS1003 detected. Version: 0
Show that nothing is read even if the DREQ line is good (vs detected)
Please check 
#define PIN_NUM_MISO GPIO_NUM_19 	// Master Input, Slave Output
#define PIN_NUM_MOSI GPIO_NUM_23	// Master Output, Slave Input   Named Data or SDA or D1 for oled
#define PIN_NUM_CLK  GPIO_NUM_18 	// Master clock  Named SCL or SCK or D0 for oled

arlaor, 2017-11-20:
hi @karawin 
I have reviewed the connections and are consistent with those provided in the image of your publication.
And how can I solve the impace of not being able to update to the new version R3?

karawin, 2017-11-20:
You ar in R3 after the update.
Type sys.version to see
I cannot say why it doesn't work on your board.
Please details the wires from the vs1053 to the esp32 including the Vcc and gnd.
How is powered your boards? etc.

arlaor, 2017-11-21:
Hi @karawin 
powered in my board is my laptop port usb 5V
command sys.version see "low memory"

karawin, 2017-11-21:
this is the answer of the http interface.
Type sys.version on telnet or uart interface.
And please answer to all questions if you need help.
Please details the wires from the vs1053 to the esp32 including the Vcc and gnd.

arlaor, 2017-11-21:
hi @karawin 
Sorry
tonight when I'm at home, I'll try to learn how to use telnet, to be able to verify the version. I will also send an image with the connections.
The voltage for the vs1053 I'm getting directly from the wemos lolin32, using the 5V pin and GND. in my ESP8266, I have it connected in the same way and it works.
Tonight I will complement the information
Thanks

arlaor, 2017-11-24:
Hi @karawin 
 Attach Image
[image]
[image]
[image]
[image]
[image]
[image]
[image]

karawin, 2017-11-24:
First of all, do an OTA to the 0.8 release, 
On restart type:
sys.logi
sys.lcd("255")

to the uart interface, the reset.
You must see more or less

I (11) main: starting app_main()
I (11) main: RAM left: 171040
E (11) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (41) main: Partition init done...
Log level is now ESP_LOG_INFO
E (2961) vs1053: VS1053/VS1003 detected. MP3Status: 40, Version: 4
I (2961) vs1053: SCI_Status  = 0x40
I (2961) vs1053: SCI_Mode (0x4800) = 0x4802
I (2961) vs1053: SCI_ClockF = 0xB800
I (2961) vs1053: VS Version (VS1053 is 4) = 4
device: 3ffcd6d8
I2S Speed: 0
I (2971) VS1053 Patches: plugin size 6350 3175
I (2971) VS1053 Patches: plugin start: 7 1 8050 6

I (3311) VS1053 Patches: plugin size 234 117
I (3311) VS1053 Patches: plugin start: 7 1 8f00 6

volume: 64
I (3421) main: hardware initialized
I (3421) main: Hardware init done...
E (3421) main: LCD Type 255
I (3421) main: audio_output_mode 4
One of I2S=0, I2S_MERUS, DAC_BUILT_IN, PDM, VS1053
I (3421) main: Set baudrate at 115200
I (3431) main: Release 0.8, Revision 0
I (3431) main: SDK v3.0-dev-1024-g92748142-dirty
I (3441) main: Heap size: 165896
I (3441) main: t0 task: 3ffcfa44
I (3441) main: starting wifi
I (3451) wifi: wifi firmware version: 111e74d
I (3451) wifi: config NVS flash: enabled
I (3451) wifi: config nano formating: disabled
I (3481) wifi: Init dynamic tx buffer num: 32
I (3481) wifi: Init data frame dynamic rx buffer num: 32
I (3481) wifi: Init management frame dynamic rx buffer num: 32
I (3481) wifi: wifi driver task: 3ffd3a1c, prio:23, stack:4096
I (3481) wifi: Init static rx buffer num: 10
I (3491) wifi: Init dynamic rx buffer num: 32
I (3491) wifi: Init rx ampdu len mblock:7
I (3501) wifi: Init lldesc rx ampdu entry mblock:4
I (3501) wifi: wifi power manager task: 0x3ffd8e88 prio: 21 stack: 2560
E (3511) wifi: esp_wifi_disconnect 844 wifi not start
I (3511) main: connecting
I (3541) wifi: mode : sta (24:0a:c4:82:ec:60)
I (3541) main: Initialised wifi
I (3661) wifi: n:11 2, o:1 0, ap:255 255, sta:11 2, prof:1
I (4651) wifi: state: init -> auth (b0)
I (4651) wifi: state: auth -> assoc (0)
I (4661) wifi: state: assoc -> run (10)
I (4671) wifi: connected with karawin2, channel 11

DNS: 192.168.1.70  
IP: 192.168.1.210

I (4841) main: RAM left 113380
I (4841) main: uartInterfaceTask task: 3ffdc5dc
I (4841) main: clientTask task: 3ffdd044
I (4841) servers: watermark: 0x7c8  1992
I (4841) main: serversTask task: 3ffddd38
I (4841) servers: telnetServer_sock socket: 4096, errno: 0
I (4841) addon: event_ir: 3ffdf220
I (4851) servers: Webserver socket: 4097, errno: 0
I (4851) addon: event_lcd: 3ffdf4a8
I (4841) main: task_addon task: 3ffdf0b8
Init ............... Done
autostart: playing:0, currentstation:13
I (7661) wifi: pm start, type:0

Important parts are
E (2961) vs1053: VS1053/VS1003 detected. MP3Status: 40, Version: 4
I (2961) vs1053: SCI_Status  = 0x40
I (2961) vs1053: SCI_Mode (0x4800) = 0x4802
I (2961) vs1053: SCI_ClockF = 0xB800
I (2961) vs1053: VS Version (VS1053 is 4) = 4

To see if the vs1053 is fully detected.
Another part
I (3691) main: audio_output_mode 4
One of I2S=0, I2S_MERUS, DAC_BUILT_IN, PDM, VS1053
 Show that the vs1053 is selected for output.
If not, go to the setting of the web page and select it.

Important: if the vs1053 is not in version 3 or 4, the output mode fall back to 0 I2S.

arlaor, 2017-11-25:
Hi @karawin 
the version of my VS1053 is shown as 0, so I suppose it fails with your project.
My vs1053, works with your version for the esp8266 and with the version for the ESP32-Radio from edzelf

karawin, 2017-11-25:
I can only confirm that it works with your vs1053 board type
[image]

arlaor, 2017-11-25:
Hi @karawin 

I understand you, I will perform tests with a module mini live esp32.
Thanks

karawin, 2017-12-28:
news?

arlaor, 2017-12-28:
Hi @karawin 

unfortunately I have misplaced the two board mini esp-32, so I will perform tests again on the plate lolin32.
In a few moments I tell you how I'm doing with the tests

arlaor, 2017-12-28:
perform tests with another board vs1053, which is working normally with the esp8266, but the message is the same

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:2980
load:0x40078000,len:0
load:0x40078000,len:11400
entry 0x40078c38
I (33) main: starting app_main()
I (33) main: RAM left: 169848
E (33) main: Running partition type 0 subtype 17 (offset 0x001d0000)
I (63) main: Partition init done...
Log level is now ESP_LOG_ERROR
# **E (1663) vs1053: NO VS1053 detected**
E (1663) main: LCD Type 0
E (3253) main:  No vs1053 detected. Fall back to I2S mode
WIFI TRYING TO CONNECT TO SSID 1
E (3463) wifi: esp_wifi_disconnect 847 wifi not start
E (4673) event: invalid static ip
E (4683) main: 
Wifi connected

DNS: 192.168.1.1  
IP: 192.168.1.160

Init ............... Done
autostart: playing:1, currentstation:1
READY. Type help for a list of commands
##CLI.NAMESET#: 1 80 HotMixRadio (80's) mp3 128 kbps
##CLI.URLSET#: streaming.hotmix-radio.net
##CLI.PATHSET#: /hotmixradio-80-128.mp3
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 100
##CLI.ICY0#: Hotmixradio 80
##CLI.ICY3#: http:\/\/www.hotmixradio.fr\/80
##CLI.ICY4#: 80s
##CLI.ICY5#: 128
##CLI.ICY7#:  ice-samplerate=44100;ice-bitrate=128;ice-channels=2
##CLI.PLAYING#

##CLI.META#: P. LION - Happy Children

karawin, 2017-12-28:
Try the last 0.9 RELEASE with more error messages.

Detection is done with the dreq line.
Check it and check the + and gnd.

arlaor, 2017-12-28:
I have tried the same cables connected to a board esp8266 and to the board vs1053 and they work normally.
The tests in esp32, I made them with version 0.9 R1

karawin, 2017-12-28:
if (VS1053_checkDREQ() == 0)
	{
		vsVersion = 0; 
		ESP_LOGE(TAG,"NO VS1053 detected");
		return;
	} 

with
uint8_t VS1053_checkDREQ() {
	return gpio_get_level(PIN_NUM_DREQ);
}
#define PIN_NUM_DREQ GPIO_NUM_34

And i am sure it works. So no other idea

arlaor, 2017-12-28:
I had saved my multimeter for a while with the batteries and these have been damaged and damaged the multimeter, so I do not have to exactly measure the voltage of pin 34.
I used a led connected to GND and pin 34 and it turns on, but I do not know exactly what voltage is receiving,
I only know that it is receiving voltage

karawin, 2017-12-29:
this code is extracted from the .9 release.

arlaor, 2017-12-30:
Hi @karawin 

precisely that board lolin 32 has some error. The test I did was to change the feed from the 5v pin of the lolin32 to the 3v pin and the detected vs1053 message appears, but a few moments later it stops recognizing it.
So I took another Lolin32 board and it worked correctly.
I thought that because WEMOS is the manufacturer, I would get a better product, but this brand is not a guarantee, unfortunately.
Thank you

karawin, 2018-01-12:
Could you close it?

arlaor, 2018-01-12:
Hi @karawin
Yes i do,  thank you

closed 2018-01-12 by karawin