karawin/Ka-Radio32 #6: DHCP Hostname

RobinMeis, 2017-11-25:
Hello Jean-Pierre,
I just finished a prototype with ESP32. Great work, it works much smoother than the ESP8266 version. However it seems, that the DHCP client is sending espressif as hostname instead of wifiwebradio.
It would be also great, if you could enhance the network setup to allow custom hostnames. This way multiple Ka-Radios could be in the same network.

Robin

arlaor, 2017-11-26:
Hi @karawin @RobinMeis 

Also, if possible, since the project allows to connect to two network SSIDs, to be able to access each SSID, a different IP, subnet mask and gateway, due to the fact that when using it in different networks, when changing the dhcp information from one network to another, when you return to the first network, you lose the connection you had for it.

karawin, 2017-11-26:
It is an old request on esp8266.
It will be done in KaRadio32.
In the meanwhile I commented this:
 If you need to have two AP, the best choice is to use the dhcp. But the ip may be difficult to predict, so i recommend to configure a static lease on your router to force dhcp to choose your preferred ip. To do that, I added the mac address of the esp. See the manual of your router about dhcp static lease.

karawin, 2017-11-26:
For the hostname espressif, is it a problem? The mad address is enough no?
The ip is allocated to a mac addr, not to a hostname.

karawin, 2017-12-10:
Good now?

closed 2017-12-11 by karawin