karawin/Ka-Radio32 #7: Ka-Radio32 firmware upload problem using ESPTOOL on Mac

gerardwr, 2017-11-26:
Hi,

I would like to try Ka-Radio 32 on my DOIT ESP32 DEVKIT V1 board, but hit a problem.

I downloaded:
http://karadio.karawin.fr/KaRadio32.bin 
and
http://karadio.karawin.fr/KaRadio32sup.zip 

Since I'm on a Mac I cannot use the Windows flasher so I used ESPTOOL:
```
imac-van-gerard:ESP32Firmware gerard$ ./esptool -p /dev/cu.SLAB_USBtoUART --baud 921600 erase_flash
imac-van-gerard:ESP32Firmware gerard$ ./esptool -p /dev/cu.SLAB_USBtoUART --baud 921600 write_flash 0x1000 KaRadio32/bootloader.bin 
imac-van-gerard:ESP32Firmware gerard$ ./esptool -p /dev/cu.SLAB_USBtoUART --baud 921600 write_flash 0x8000 KaRadio32/partitions.bin 
imac-van-gerard:ESP32Firmware gerard$ ./esptool -p /dev/cu.SLAB_USBtoUART --baud 921600 write_flash 0x10000 KaRadio32/KaRadio32.bin 
imac-van-gerard:ESP32Firmware gerard$ ./esptool -p /dev/cu.SLAB_USBtoUART --baud 921600 write_flash 0x1d000 KaRadio32/KaRadio32.bin 
```
The uploads are succesfull.

When I reset the ESP I get from the serial port:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:2980
load:0x40078000,len:0
load:0x40078000,len:11400
entry 0x40078c38
E (120) esp_image: invalid segment length 0x6f742064
E (120) boot: OTA app partition slot 0 is not bootable
E (120) esp_image: image at 0x1d0000 has invalid magic byte
E (124) boot: OTA app partition slot 1 is not bootable
E (129) boot: No bootable app partitions in the partition table
user code done
```

Did I make a mistake in uploading, or do you have any other suggestions what to try?

gerardwr, 2017-11-26:
Solved the problem.

The last command (upload to 0x1d0000) seems to cause the problem. I started again, but dit NOT execute the last command. Now the output is:
````
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:2980
load:0x40078000,len:0
load:0x40078000,len:11400
entry 0x40078c38
I (12) main: starting app_main()
I (12) main: RAM left: 170528
E (12) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (52) main: Partition init done...
Log level is now ESP_LOG_ERROR
E (1652) vs1053: NO VS1053 detected
E (1652) main: LCD Type 0
E (3692) main: The default AP is  WifiKaRadio. Connect your wifi to it.
Then connect a webbrowser to 192.168.4.1 and go to Setting
May be long to load the first time.Be patient.
Init ............... Done
autostart: playing:0, currentstation:0
```

gerardwr, 2017-11-26:
Works excellent, Thanks for sharing.

karawin, 2017-11-27:
Yes you forgot a 0 for 1D0000

closed 2017-11-27 by karawin

gerardwr, 2017-11-27:
Checked it several times, stupid me, my bad! Thx.