karawin/Ka-Radio32 #9: ssd1331

hopla123, 2017-11-29:
Hi,
I have issue with rotated text by display SSD1331 (looks like 90 degree rotation), can you help me? 
Regards
Jan
[image]

karawin, 2017-12-10:
I will see this point soon.

karawin, 2017-12-10:
Do an OTA update and tell me if you need a 180° rotation.

hopla123, 2017-12-10:
Thank you about help, now is rotation ok, only some additional text is not fully in display size.
[image]

karawin, 2017-12-10:
Yes i have to adjust for this small screen.
Wait ;-)

karawin, 2017-12-10:
Try again.... now

hopla123, 2017-12-11:
Hi, 
now is text inside lcd but text fonts looks as cut of.
Regards
Jan
[image]

closed 2017-12-11 by hopla123

karawin, 2017-12-11:
again... test it aftezr ota

hopla123, 2017-12-11:
ok, now looks fine, only startup menu is too large.
[image]
[image]