karawin/Ka-Radio32 #12: New feature requests

sparx266, 2017-12-17:
Awesome radio project here!

My infra red remote is doing funny things but is working, will have to look at that.
Would be great to have a power off/on via IR, but what would be brilliant is to have MP3 files either from SD card or USB stick via a host board playable as well as the radio.

Bluetooth speaker would be really cool too.

Once again, this is great, thanks for doing it.

karawin, 2017-12-19:
Thanks,
What is a power off/on from the software point of view?
I can ask the processor to go to sleep, but cannot control lcd, vs1053 etc power.
Sure it can read from a sd card, but we are in 2017. mount a local SHOUTcast server and make your own radio. A must.
Bluetooth speaker: I tried but not enough sram for both.

marcinmaster999, 2017-12-20:
ok. a2dp source as server bluetooth?

karawin, 2017-12-20:
Good try but...
https://github.com/espressif/esp-idf/issues/580
;-)

sparx266, 2017-12-20:
Fair point, what about a standby mode then?  Display could show the time?  VS1053 should be silent, volume 0?

Not sure a local shoutcast server is for me.  Reading MP3's off a USB drive would be better, ie more convenient for playing locally stored MP3's.  But most/many VS1053 boards already have an SD card slot built in?

Pity about the Bluetooth speaker, is there any way around that?

ARMOR888, 2019-01-21:
Can you add the ability to read playlists on the network without specifying the port? I have a network storage with music. I make a playlist accessible via the link \\192.168.1.1\MyNAS\Music\MyPlaylist.m3u This link succesfully opens in a music player, for example MPC-HC or VLC, but KaRadio can't open this link right now. It would be very very nice if i could open and save not only radio stations, but also such playlists in the same station list. In truth, this is exactly what I need to listen to my music library without turn on large devices like SmartTV or PC.

karawin, 2019-01-22:
'' Can you add the ability to read playlists on the network without specifying the port? I have a network storage with music. I make a playlist accessible via the link \192.168.1.1\MyNAS\Music\MyPlaylist.m3u ''
It is not a port problem but a network protocol problem. Your pc is not an html server and smb is not implemented in esp32.
Another solution is to write a little soft to convert the playlist in a txt file uinderstandable by KaRadio.
But it doesn't work for local pc files. They must be on a html server.

karawin, 2019-01-22:
Done a while ago.
The list of format is
				char *t0 = strstr(clientPath, ".m3u");
				if (t0 == NULL)  t0 = strstr(clientPath, ".pls");
				if (t0 == NULL)  t0 = strstr(clientPath, ".xspf");				
				if (t0 == NULL)  t0 = strstr(clientPath, ".m3u8");	
				if (t0 == NULL)  t0 = strstr(clientPath, ".asx");					
				if (t0 != NULL)  // a playlist asked

emax73, 2019-01-22:
Probably, small need "import/export m3u8" button at Edit tab of web page

karawin, 2019-01-22:
The web page is only a cached view of the esp32 data with some facilities.
- If done in the javascript, it is only available on the web page as  instant play feature. seen from the esp32.
May be the less worst choice for some little playlists.
- If sent to the esp32, it replace the current list that can be restored by re reading the txt file.
- or converted in txt KaRadio format and treated like any other saved files.
All these are not very clean.

emax73, 2019-01-22:
Now I import m3u8 playlist made for AIMP3 windows player by custom php script, 
I see, that jet one man created php script for convert m3u8 playlist to WebStations.txt internal KaRadio32 format

karawin, 2019-01-22:
The internal format is a simple json text
{"Name":"RTL","URL":"streaming.radio.rtl.fr","File":"/rtl-1-48-192","Port":"80","ovol":"-4"}
For historical reason, the path is named File. ovol is the relative offset volume.

No php allowed (it must run on the esp32 for that, but...). Just javascript on the browser.

emax73, 2019-01-22:
PHP selected as well known language and run once at PC,
May be any open source javascript parser with convert to internal json, or RegularExpression js parser?

emax73, 2019-01-22:
I have php server at my PC

karawin, 2019-01-22:
and?
Explain what you want to do.

karawin, 2019-01-22:
It must be an easy solution for all users

emax73, 2019-01-22:
Usefull feature will been take well known m3u8 playlist and loaded to KaRadio32 station lists instread of internal not using nowhere WebStations.txt

karawin, 2019-01-22:
Ok do it

emax73, 2019-01-22:
Its optional suggestion, will can been useful at your free time

karawin, 2019-01-22:
;-) But we loose the ovol.

emax73, 2019-01-22:
ovol can still at WebStations.txt, at m3u8 ovol = 0 unfortunally

karawin, 2019-01-22:
ARMOR888 I don't think. Probably a mp3 file on a sd card.

Tonymac32, 2019-05-05:
I've got a board I'm launching that uses the RPi form factor.  This gives me a unique situation where I2S needs to be on alternate pins to hardware DAC.  No issue there, I can assign the pins and it works beautifully, then select the DAC and it works too, from the TRS jack in the expected RPi position.  The issue is trying to use PDM in this configuration, it comes out the I2S pins of course.  

Is it possible to allow for the definition of a "P_PDM_L" and "P_PDM_R" pins in the hardware file?  Since reboot is necessary between audio outlets, this would support situations where an I2S and an analog both exist in one system.

https://twitter.com/TonyMac_32/status/1124079867828559872

karawin, 2019-05-05:
Not sure.
I wonder if DAC and PCM can be configured with GPIO not equal to (GPIO25 & GPIO26)
It works for I2S but for DAC or PCM the pin's config must be :
pin: I2S Pin structure, or NULL to set 2-channel 8-bit internal DAC pin configuration (GPIO25 & GPIO26)
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/i2s.html

Tonymac32, 2019-05-05:
The onboard DAC works irrespective of pin assignments (analog output to
pins 25 and 26 no matter what, I could compare quality though just in
case).  PDM comes through the defined I2S pins, being an I2S mode, which is
my only real issue, hence the thought to be able to define the I2S pins
differently for that mode (and maybe onboard DAC as well).

karawin, 2019-05-05:
Checked.
Only DAC output are static GPIO's.
I will ad your suggestion in next release
 "P_PDM_L" and "P_PDM_R"

karawin, 2019-05-08:
Checked the software. The pdm pin are wrongly not assigned in the source. So output are in the default pins.
I can change the soft to use the i2s definitions for i2s and pdm output.
So "P_PDM_L" and "P_PDM_R" are not needed. Only the mode is enough.
For the pdm mode, i think that the output is PDM_CLK and PDM_DATA like
[image]

Is it ok or am i wrong?

Tonymac32, 2019-05-09:
Hmmm, that explains some odd behaviors I was seeing, like only getting output on "warm" reset and not on a full power cycle...

As to the format of the output, I will have to do some checking, I don't think any demuxing was necessary.

Tonymac32, 2019-05-09:
OK, for Mono output no demuxing is required (only LPF), but for stereo it is, which requires a proper amplifier to handle the data/clock signal.  My mistake, the documentation is not the easiest to find...  Some info on the topic:
https://translate.google.com/translate?hl=en&sl=ja&u=http://blog-yama.a-quest.com/%3Feid%3D970192&prev=search

I agree splitting the paths doesn't make a lot of sense at that point, it can be kept with the I2S

karawin, 2019-05-14:
You can try the sys.prerelease available.

bazooka07, 2020-01-17:
For importing playlists in .m3u or .pls format follow this link :
https://bazooka07.github.io/Ka-Radio32/