karawin/Ka-Radio32 #14: Integration request

WhizzCat, 2018-01-04:
Hi, love your project, first of all!

This is, of course, not really an issue, but more of a feature request. I've been dabbling with the folks at https://www.blynk.cc/ This is one awesome IoT solution.

I'm wondering how hard it would be to integrate this into your amazing Radio solution. I'm not really a coder, so I'm a little hesitant as to making guesses about how you programmed all your stuff. But since the ESP32 has two cores, would it be possible to run the Blynk solution on one core? It requires somewhat big computing time and cannot be mixed with blocking code since it's realtime.

There is an example piece of how this basically works over here: https://examples.blynk.cc/?board=Raspberry%20Pi&shield=System%20default&example=GettingStarted%2FBlynkBlink

In essence, it's a piece of middleware connecting to both an App on your Phone and a Server (whether it be Cloud or a Local server you run on your own time).

Or maybe you can give me a hint how to integrate this and if it's possible to run it on a dedicated core on the Radio.

Thanks very much in advance and if you need more info, I'd be happy to help out :)

Greetings, Bastiaan (from the Netherlands).

karawin, 2018-01-12:
Not really a problem of core. A simple task on freertos, but is iot very useful for a webradio?
You can control it with a rotary, an uart interface, a telnet interface, a web interface. Why another one
 I admit I don't like IOT.

WhizzCat, 2018-01-13:
That is, of course, your prerogative. I'll just try and stick to the web commands than. It requires me to do a little more work, but that's not really an issue. Thank you for responding!

closed 2019-02-16 by karawin