karawin/Ka-Radio32 #16: New feature suggestion : Touch control of Ka Radio 32

gerardwr, 2018-01-08:
This is a suggestion for an additional feature to use the ESP32 "touch" feature in Ka Radio 32.

Now that I have attached an Oled display to my KaRadio-32 I think it would be great to control Karadio without using the webpage.

The use of the “ESP32 touch” feature to control Karadio32 would be a nice addition, ref:
http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/touch_pad.html

I have seen an addon with mechanical microswitches  for controls (the ESP8266 version?) so maybe the basic code is in place and adding touch control is not too complicated.

Looking at my Oled SSD1306 display the use of the metallic islands of the mounting holes would make excellent touch-controls. As far as I can see the islands are not connected to the rest of the PCB. It would require:
- 4 additional wires from the mounting holes to the ESP32 touch GPIO's
- and ofcourse software changes

[image]

karawin, 2018-01-12:
Hemm... buy a ky040 rotary encoder. Much more easy and a full control in one.

gerardwr, 2018-01-12:
Ok, I understand.

But I must say I like the very simple basic setup with just the ESP32 and the Oled and no extra hardware like an external DAC, external decoder, etc.

Not a real problem for me, In practice I always listen to the Frech Morow Radio station :-)

karawin, 2018-01-12:
Play/Stop        Volume/Station

vol/station -    vol/station +

Is better no?
Interesting project but not generic
So I will not do it because of the lack of available gpio

gerardwr, 2018-01-12:
Yes, your suggestion for buttons is beter.

I understand your reason for not implementing it. no hard feelings.

Thx for the great software, Closing this issue now.

closed 2018-01-12 by gerardwr