karawin/Ka-Radio32 #17: Freature request : Restart Wifi connection after connection is lost

gerardwr, 2018-01-10:
Hi,

Would it be possible to restart the Karadio32 Wifi connection to the Wifi router when the connection was lost?

Background:
Karadadio is very stable for me. It often plays a station for < 24 hours without any issue.

But sometimes Karadio32 looses the Wifi connection. On the LCD a.o. the time is displayed in a BIG font, instead of the Radio details. If you try to connect to the webpage it does not connect. So my conclusion is that the software is still running but that the Wifi connection was lost for some reason.

My Wifi signal is not perfect, so that may be the reason why Kardaio32 was disconnected from the Wifi router.

Would be great if Karadio would restart the Wifi connection is the connection was lost for some reason.

Thx for an excellent piece of software!

karawin, 2018-01-12:
The wifi connection is restarted but the station play is not even with autostart.. I will check it on reconnection.

gerardwr, 2018-01-12:
In my case the connection was really lost. The ESP32 was not connected to my network anymore, and connection to the webpage of the ESP32 was not possibly anymore.

After restarting the ESP32 the radio  station was autoplayed again.

gerardwr, 2018-01-12:
Yes that’s what  I have to do now.

Would be much better if Karadio would do it automatically, So it would play “forever” even after a connection problem.

gerardwr, 2018-01-13:
What's weird : Generally a radio station plays withouthout interruption for > 24 hours. But then:
- the stream stops
- the oled shows Date/Time (is running)/IP
- The webpage shows "invalid address" while it has been playing the address for hours:
[image]

- When I then press "play" on the webpage the stream does NOT start again and the oled displays says "stopped".
- When I reset the ESP the stream starts playing automatically.

So it seams that:
- Karadio is still running
- connection from browser to ESP32 is still running
- Karadio has lost connection to the stream and is not able to restart it.

karawin, 2018-01-13:
Invalid address: no answer to a connection request. I think I have to write "not reachable" in place.

When a station plays, and the stream stops. After a short tempo the connection is relaunched. If you see the log, you will see "try restart". But if the http GET request don't receive an answer (no tcp connection), the invalid address is shown and KaRadio go to the stop state and stay in it.
In stop state, after a delay, the lcd shows the time.
A manual Start is required. Hard to differentiate a wrong address from a transient failure.
May be i have to remind if the connection was on, but actually it is not done.

gerardwr, 2018-01-13:
@karawin 
Thx for your explanation how Karadio handles a shaky connection. I must admit I did not have a look on the Karadio source. I'm still struggling with installing ESP32-IDF :-(

I think this is a nice subject for adding to the README, I can write a proposal for it, is that OK?

Would it not be a good idea to try to restart after displaying the time for a little while, like 5 mins. Streams often have temporary hickups.

gerardwr, 2018-01-13:
Here the proposal to add to the README

---------------

## RADIO STREAMS
If Karadio32 is configured to use ADC, PDM or I2S output (to an external dac) it can play audio streams in mp3 format. Other formats AAC, WMV, etc. Require too much processing power and can not be played this way.

When you want to play other formats you add an external VS1053 decoder to Karadio. Details on the features of this decoder can be found here : http://www.vlsi.fi/en/products/vs1053.html

### HANDLING STREAMING ERRORS

Normally a radio station is played without interruption.

But when a connection to the server of the radio station is (temporarily) disrupted (due to internet problems or server problems) Karadio behaves as follows:
- Playing of the radio stream stops, after a short while Karadio retries to relaunch the connection to the radio stream.
- If the relaunch of the connection fails again, the webpage of Karadio then shows “invalid address” and goes into the stop state. If you have a display connected to Karadio it will then show the Date, Time and IP address.
- You can try to relaunch the connection manually by pressing “play” on the Karadio webpage, or resetting the ESP32.

karawin, 2018-01-13:
Great. I will include it.
Remark: AAC not playing is not a problem of processing, but only a problem of lack of sram.

karawin, 2018-01-14:
On 0.9 R4, the esp32 reboots when the wifi is disconnected.

gerardwr, 2018-01-14:
That’s excellent. I Will update immediately, thank you for this quick fix.

gerardwr, 2018-01-14:
Feature was implemented as requested, So this issue is closed.
Thx.

closed 2018-01-14 by gerardwr

MasterYura, 2020-11-17:
I have sometimes error with one station: when I click start or change station - karadio32 return: "invalid address"
But this station works fine a few second ago. and also this station works fine in Chrome and Aimp at the same time (the same ulr)
Wifi in esp32 is also ok, because another station play fine at the same time
what it can be ?

Settings: Output: DAC or I2S (currenty only testing by DAC connected directly to speakers
device: ESP32 Wrover https://aliexpress.ru/item/4001006415969.html?spm=a2g0s.12269583.0.0.fa7e389aNmuAVQ (color wrover)
[image]

KaRadio32 V1.9 R7
esp32 is connected to pc without any another connection (cables)

video of error with log  https://drive.google.com/file/d/12XoOhun8tmFL1Ip2TsV5e4PvceKzNcNA/view?usp=sharing