karawin/Ka-Radio32 #18: HW Support Request: WiFi LoRa 32 Board from HelTec Automation

gvoigtlaender, 2018-01-10:
Hi,

so far Ka-Radio32 runs well on Heltec LoRa32 Wifo Kit, just the OLED uses different IO. Seems to be using some pins which are currently reserved for ENC & LED. Any change to get that working? 

Please find further info here:
https://robotzero.one/heltec-wifi-kit-32/
https://hackaday.io/project/26991-esp32-board-wifi-lora-32

BR,
georg

karawin, 2018-01-12:
See the Ka-Radio32\main\include\gpio.h file and change with your needs.
And yes, you need to generate KaRadio by yourself.
I try to be as generic as possible but for these special requests, the best is to create 
Thanks for your interest.

gvoigtlaender, 2018-01-12:
I'm on it, still checking which packages are missing in my gentoo

karawin, 2018-01-12:
https://esp-idf.readthedocs.io/en/latest/get-started/linux-setup.html

gvoigtlaender, 2018-01-12:
Thx, getting closer. I was missing the xxd command (vim-package)

Now I get stuck at 
>~/esp/Ka-Radio32/main/./addonucg.c:11:25: fatal error: driver\gpio.h: No such file or directory

gvoigtlaender, 2018-01-12:
Solved: 

addonucg.c uses wrong backslash in include, needs to be slash ;)

Now i'll check the gpio.h

karawin, 2018-01-12:
Yes I will correct it. Msys32 is more tolerant than a real linux.

gvoigtlaender, 2018-01-12:
addonu8g2.c:445 PIN_LCD_RST -> PIN_I2C_LCD

karawin, 2018-01-12:
u8g2_esp32_hal.reset = PIN_I2C_RST;

Thanks

gvoigtlaender, 2018-01-14:
Hi,

patching the addonu8g2.c as above and adjusting gpio.h made the OLED working

>#define PIN_I2C_SCL GPIO_NUM_15
>#define PIN_I2C_SDA GPIO_NUM_4
>#define PIN_I2C_RST	GPIO_NUM_16

Thanks,
georg

closed 2018-01-14 by gvoigtlaender