karawin/Ka-Radio32 #19: Suggestion : Add Audio Output section to README

gerardwr, 2018-01-10:
Hi,

While experimenting I created some extra description for "Audio output", maybe you can use it for the README.

Thx.

------------

# Audio output

In the Setting panel on the webpage of KaraDio32 you can set the desired output method for audio.

For output on a simple speaker or a general analog amplifier
- DAC to use the built in DAC of the esp32
- PDM to output a PDM (Pulse Density Modulation) stream. Wiki on PDM : https://en.wikipedia.org/wiki/Pulse-density_modulation

For output via addional I2S or VS1053 hardware
- I2S for connection to ac external DAC
- I2SMERUS to connect a merus amplifier
- VS1053 to connect to a vs1053 board.

### Connecting a speaker, earphone or amplifier with DAC or PDM setting
You can connect the GPIO25 (left audio channel), GPIO26 (right audio channel) and Ground directly to a small loudspeaker, a simple earphone (like from an mp3 player) or the input of an analog amplifier. The quality is less perfect than using an I2S or VS1053 board, bit it is very simple hardware-wise.

Connect like this
```
                         +------------------------------------+
ESP32-GPIO25 (left) -----+speaker/earphone/analog-amp (left)   +-----+
                         +------------------------------------+      |
                                                                     |
                         +------------------------------------+      |
ESP32-GPIO26 (right) ----+speaker/earphone/analog-amp (right  +------+
                         +------------------------------------+      |
                                                                     |
ESP32-GROUND --------------------------------------------------------+
```

### Reducing the analog output level
If the audio signal level is too high for your speaker, earphone or amplifier you can add a potentiometer to decrease the level. Connect the potentiometer like this (only pictured for 1 audio channel, for stereo you need 2 potentiometers or a stereo-potmeter.

```
                                     +--------------------------------------+
ESP32-GPIO25 (left) -------+   +-----+speaker/earphone/analog-amp (left)    +------+
                          +-+  |     +--------------------------------------+      |
                          +P+  |                                                   |
                          +o+--+                                                   |
                          +t+                                                      |
                          +++                                                      |
                           |                                                       |
ESP32-GROUND --------------+-------------------------------------------------------+
```
### Improving the audio quality with a filter
An analog DAC signal always has some noise that may cause some distortion on the audio output, especially on low volume passages in the sound. This noise can be decreased with a low-pass filter. The digital PDM signal needs allways low pass filter to convert the digital signal to an analog signal. Fortunately the speaker and earphone acts as a low pass filter, although not in a perfect way.

You can improve the analog signal with an external low-pass filter. A simple low-pass RC filter can be find on the internet, f.i. here : http://www.pavouk.org/hw/usbdac/en_index.html.

karawin, 2018-01-12:
Great contribution, I will add it.
Thanks

gerardwr, 2018-01-13:
Is added to readme, closing.

closed 2018-01-13 by gerardwr