karawin/Ka-Radio32 #21: NEC ir setup

phanus, 2018-01-28:
First of all, thanks for awesome project and all of the hard work!
I'd like to ask, how to set up the remote control? If I enable log on debug level, I see successfully detected codes of NEC remote control keys in serial terminal. How to link them to certain actions like Volume + or - , switch to station x, etc.?
Thank you.

karawin, 2018-02-02:
See addon.c function void irLoop()
You have to add your own codes for your remote.

phanus, 2018-02-10:
Perfect! This is a piece of code I was looking for. Thank you.
BTW It would be super convenient to have the IR codes configurable over the web interface. Adding new codes to the current list increases risk of collision with some appliance you may already have. Like the TV...

karawin, 2018-02-11:
This is a very static configuration, not really useful on the web interface, but may be on the uart interface.

phanus, 2018-02-11:
Indeed. Similar to display configuration - you do it once at the beginning and then you can forget about it. UART would do the job.

yura174, 2018-07-03:
help the console gives out
I (991986) addon: IR event: Channel: 0, ADDR: 206e, CMD: 7f2 = 206EF2, REPEAT: 0
I (993936) addon: IR event: Channel: 0, ADDR: 206e, CMD: bf2 = 206EF2, REPEAT: 0
I (994606) addon: IR event: Channel: 0, ADDR: 206e, CMD: ff2 = 206EF2, REPEAT: 0
I (995276) addon: IR event: Channel: 0, ADDR: 206e, CMD: 1bf2 = 206EF2, REPEAT: 0
I (995876) addon: IR event: Channel: 0, ADDR: 206e, CMD: 17f2 = 206EF2, REPEAT: 0
I (996546) addon: IR event: Channel: 0, ADDR: 206e, CMD: 13f2 = 206EF2, REPEAT: 0
I (996846) addon: IR event: Channel: 0, ADDR: 206e, CMD: 23f2 = 206EF2, REPEAT: 0

closed 2018-11-26 by karawin