karawin/Ka-Radio32 #23: Feature Request: Hardware Button Control

chellmann, 2018-02-09:
Hi,

it would be great to select the stations with dedicated Buttons. For example in combination with a MPR121 breakout-board which has a I2C Interface and 12 possible inputs, even with capacitive sensing.
The rotary encoder is great to control the volume, but to switch between the first stations in the list it would be much easier with simple buttons - like on the good old car radios.

MarcLuc, 2018-02-10:
I thank this is a good idea. Having presents station button or favourites button is a common feature on radios.

karawin, 2018-02-11:
Yes sure, it is one of many solutions.
Rather than including this feature in the code, i think it is better to think about an external addon with a stm32 or other slow arduino cpu and a command with the serial interface.

karawin, 2018-11-26:
Resolved since 1.4. I close it

closed 2018-11-26 by karawin