karawin/Ka-Radio32 #24: Utilise esp32 bluetooth as another audio output/control for Karadio

MarcLuc, 2018-02-10:
Another feature worth looking into is using bluetooth wireless headphones as another possible audio output and playback control?   bt headphones like sony wireless headphones MDR-ZX770BN... That way user/ listener can adjust volume or channel via BT headphones.   Let me know if you need someone to test!

karawin, 2018-02-11:
I can't. Not enough sram to include BT in code.
I will try withe the esp32 WROVER.

DennisSc, 2018-07-16:
I am currently looking into A2DP SRC capabilities as well (ESP32 as bluetooth playback device). Maybe you could use an external SRAM chip, like [IPS6404L-SQ-SPN](https://www.electrodragon.com/product/2pcs-ipus-ips6404-iot-ram/) for example? See https://esp32.com/viewtopic.php?t=5450

karawin, 2018-12-24:
It is in progress.
A2DP sink and source for wrover esp32
Planned in 2019....

For pure A2DP sink, some other project are available yet on wroom

fintros, 2019-09-21:
Any progress so far?