karawin/Ka-Radio32 #25: Ka-Radio32 crash while changing presets via UART

phanus, 2018-02-10:
I made a small board with Arduino and IR receiver and couple of buttons for favorite preset selection.
After sending a few commands to the Radio like Serial.print("cli.play(\"5\")\r"); or Serial.print("cli.next\r"); the Radio crashes with following error.
It crashes also when sending the commands via serial console from PC.

`
***ERROR*** A stack overflow in task uartInterfaceTa has been detected.
abort() was called at PC 0x4008b3c0 on core 1

Backtrace: 0x4008b2de:0x3ffdc0e0 0x4008b3a7:0x3ffdc100 0x4008b3c0:0x3ffdc120 0x400877b8:0x3ffdc140 0x4008966c:0x3ffdc160 0x40089622:0x3ffdc9f0

Entering gdb stub now.
$T0b#e6+$#00
`

karawin, 2018-02-11:
I will increase the stack size of this task.
app_main.c line 926
xTaskCreate(uartInterfaceTask, "uartInterfaceTask", 2200, NULL, 2, &pxCreatedTask); 
to
xTaskCreate(uartInterfaceTask, "uartInterfaceTask", 2400, NULL, 2, &pxCreatedTask);

phanus, 2018-02-19:
Fixed in 0.9 R5. Works nicely. Thank you!

closed 2018-02-19 by phanus