karawin/Ka-Radio32 #26: ESP32-COLOR-ADB hangs when SSD1331 was enabled

makserge, 2018-03-06:
Hello.
I have a trouble with ESP32-COLOR-ADB from Microwavemont (tindie.com).
This board have PCM5102 DAC and SSD1331 OLED on board.

I checked out this project and flashed it this board with default configuration.
Opened web interface and setup wifi connection and loaded my playlist.
After reboot I was able to connect to web interface and play radio.
OLED was disabled at all.

Than I tried to enable OLED by changing SPI pins for OLED

in main/include/gpio.h this way

@@ -48,9 +48,9 @@
  // SPI lcd
  //---------
  // HSPI pins +
 -#define PIN_LCD_CS	GPIO_NUM_13		//CS
 -#define PIN_LCD_A0	GPIO_NUM_14		//A0 or D/C
 -#define PIN_LCD_RST	GPIO_NUM_2		//Reset RES RST or not used
 +#define PIN_LCD_CS	GPIO_NUM_5		//CS
 +#define PIN_LCD_A0	GPIO_NUM_16		//A0 or D/C
 +#define PIN_LCD_RST	GPIO_NUM_17		//Reset RES RST or not used

and flash updated code to the board

Than I set current LCD type via telnet 

sys.lcd("197");

and reboot

OLED shows some info and blinking once in 2 seconds.

Board can't connect to the wifi and doesn't respond to web requests, telnet and even ping to its IP.

Could you help with issue?
Should I change something else to enable OLED on this board?
Is my OLED pins configuration OK?

Board pin assignment is here
https://cdn.tindiemedia.com/images/resize/qOo86qvwCBgHBnt77bsAibHUS6k=/p/fit-in/1300x866/filters:fill(fff)/i/73055/products/2018-02-08T01%3A02%3A33.164Z-ESP32-ad2.png

Another question, can I change default LCD type to 197 instead of default 0?
Where is right place in code to set this type?

karawin, 2018-03-11:
I need the log of the uart.
The default lcd is kept on the config. No need to change it at each reboot.

makserge, 2018-03-12:
How can I enable logs in UART? 
Changed log level via menuconfig and reflashed ESP -> nothing in UART console (used Arduino Serial Monitor)

My current gpio.h doesn't have repeating pin definitions

#pragma once
#ifndef __GPIO_H__
#define __GPIO_H__

//-------------------------------//
// Define GPIO used in KaRadio32 //
//-------------------------------//
// Compatible ESP32 ADB
// https://www.tindie.com/products/microwavemont/esp32-audio-developing-board-esp32-adb/

// HSPI pins
//-----------
#define PIN_NUM_MISO GPIO_NUM_19 	// Master Input, Slave Output
#define PIN_NUM_MOSI GPIO_NUM_23	// Master Output, Slave Input   Named Data or SDA or D1 for oled
#define PIN_NUM_CLK  GPIO_NUM_18 	// Master clock  Named SCL or SCK or D0 for oled

// status led if any.
//------------------- 
//Set the right one with command sys.led
// GPIO can be changed with command sys.ledgpio("x")
#define GPIO_LED	GPIO_NUM_12		// Flashing led or Playing led

// gpio of the vs1053
//-------------------
#define PIN_NUM_XCS  GPIO_NUM_32
#define PIN_NUM_RST  GPIO_NUM_12 
#define PIN_NUM_XDCS GPIO_NUM_33
#define PIN_NUM_DREQ GPIO_NUM_34
// + HSPI pins

// Encoder knob
//-------------
#define PIN_ENC_A   GPIO_NUM_35		// CLK
#define PIN_ENC_B   GPIO_NUM_14		// DT
#define PIN_ENC_BTN GPIO_NUM_2		// SW

// I2C lcd (and rda5807 if lcd is i2c or LCD_NONE)
//------------------------------------------------
#define PIN_I2C_SCL GPIO_NUM_5
#define PIN_I2C_SDA GPIO_NUM_16
#define PIN_I2C_RST	GPIO_NUM_17		// or not used

// I2C rda5807 (if lcd is spi)
//----------------------------
#define PIN_SI2C_SCL GPIO_NUM_15
#define PIN_SI2C_SDA GPIO_NUM_27

// SPI lcd
//---------
// HSPI pins +
#define PIN_LCD_CS	GPIO_NUM_5		//CS
#define PIN_LCD_A0	GPIO_NUM_16		//A0 or D/C
#define PIN_LCD_RST	GPIO_NUM_17		//Reset RES RST or not used

// IR Signal
//-----------
#define PIN_IR_SIGNAL GPIO_NUM_21	// Remote IR source

// I2S DAC or PDM output
//-----------------------
#define PIN_I2S_LRCK GPIO_NUM_25	// or Channel1
#define PIN_I2S_BCLK GPIO_NUM_26	// or channel2
#define PIN_I2S_DATA GPIO_NUM_22

// to set a value: 		gpio_set_level(gpio_num_t gpio_num,value);
// to read a value: 	int gpio_get_level(gpio_num_t gpio_num)

void gpio_output_conf(gpio_num_t gpio);

#endif

gojimmypi, 2019-06-10:
Is `VSPI` actually pins `//SCLK = 18, MISO = 19, MOSI = 23, SS = 5` 
and `HSPI` pins `//SCLK = 14, MISO = 12, MOSI = 13, SS = 15`
?

See [this example](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino#L37).

I ask, as my `VSPI` is unavailable (different board: WROOM-32), and I am [unable](https://github.com/emard/ulx3s/issues/8) to get `HSPI` working; I'd like the performance of HWSPI, and not do software SPI (which I _can_ get working, but is considerably slower)

Thanks in advance for any word of wisdom on this topic. :)