karawin/Ka-Radio32 #27: M5Stack ILI9341 rather not work

Kabron287, 2018-03-10:
I configured addon.h and gpio.h as following:
`// lcd_type
// min 128x64
#define LCD_COLOR		128	//0x80
#define LCD_SPI			194 //0x40
#define LCD_I2C			0
#define LCD_NONE		255 `

`// SPI lcd
//---------
// HSPI pins +
#define PIN_LCD_CS	GPIO_NUM_14		//CS
#define PIN_LCD_A0	GPIO_NUM_27		//A0 or D/C
#define PIN_LCD_RST	GPIO_NUM_33		//Reset RES RST or not used`

compiled OK, stations receive OK
sys.lcd                                                                                                                             
##LCD is 194#
but after start I see very very slowly LCD filling with light blue and nothing more.

karawin, 2018-03-11:
You need to reassign gpio in conflict with the vs1053 even if not used.
Dont touch to the addon.h file.
use sys.lcd("197") to set it one time.

Kabron287, 2018-03-11:
M5STACK
I modified
**//#define PIN_NUM_XDCS GPIO_NUM_33
#define PIN_NUM_XDCS GPIO_NUM_3**
nothing changed.
WROVER 
I see no conflicts with vs1053 and LCD(OLED) initialization and primary output flows normally, but station parameters results in one pixel line. Here is my gpio.h
[gpio.h.txt](https://github.com/karawin/Ka-Radio32/files/1800423/gpio.h.txt)

closed 2018-03-11 by Kabron287

karawin, 2018-03-11:
/ SPI lcd
//---------
// HSPI pins +
#define PIN_LCD_CS	GPIO_NUM_22		//CS
#define PIN_LCD_A0	GPIO_NUM_21		//A0 or D/C
#define PIN_LCD_RST	GPIO_NUM_18		//Reset RES RST or not used

// IR Signal
//-----------
#define PIN_IR_SIGNAL GPIO_NUM_21	// Remote IR source

// I2S DAC or PDM output
//-----------------------
#define PIN_I2S_LRCK GPIO_NUM_25	// or Channel1
#define PIN_I2S_BCLK GPIO_NUM_26	// or channel2
#define PIN_I2S_DATA GPIO_NUM_22

Conflict on 21 and 22 and may be others.
Sorry no time to detail all.
Best to draw a scheme with all gpio to see the problem