karawin/Ka-Radio32 #36: Board support (TTGO ESP32 Tm Music)

joaofl, 2018-06-13:
I've been trying to port your code to this board. It includes changing a couple of IOs, configuring the proper LCD, push buttons, and It should be good. I'm specially interested on this [board](https://www.aliexpress.com/item/TTGO-ESP32-Tm-Music-albums-2-4-inch-TFT-PCM5102A-SD-WiFi-Module-Bluetooth-development-board/32848902211.html?spm=a2g0s.9042311.0.0.6e444c4d6M7uWZ). Find the schematic [here](http://www.ttgobbs.com/attachment.php?aid=ODF8NDgyMDYyYzh8MTUyODkwNzQyNXw5YmI5L05icVViVmhKd2xvdDdUZm5XcERGelY2ZWV1ejRGNUdYZ0Z0Q0ZGbzRzbw%3D%3D) It uses the PCM5102 I2S DAC.

Its the most complete ready-to-run solution I found. relatively low cost.

But now, as I'm trying to do that, I went through one issue, and one question:

- How do you think it is a good idea to have a complete board support? Instead of changing individual configs, but having a whole config-set for an specific board? I tough of having some pre-set config structures, that could be changed on the terminal (something like having **renderer_config** and **board_config** pre-sets for a given board. How does this sound? I could contribute on that. I'm quiet interested on getting this board to work smoothy.

- My other issue, which is more technical one: I've been trying for the last couple of weeks to get the audio playing fine. Issue is that I'm always getting musics playing as if they were in fast-forward, with noise as well. I tried changing sampling rate, master clock, clock sources, etc.. Nothing solves the issue.

I found a fork of the the KaRadio from which, If I flash the binaries already compiled, it works, otherwise, it does not. Really dont know where to look more.

joaofl, 2018-06-14:
> My other issue, which is more technical one: I've been trying for the last couple of weeks to get the audio playing fine. Issue is that I'm always getting musics playing as if they were in [...]

I found out why and how to solve it. Its basically the I2S library that got major updates lately. By using the latest IDF with the previous I2S library version, I managed to compile and get radio working. Now I have to figure out a way to get it working without downgrading. Need to investigate more.

karawin, 2018-06-14:
Ok, I  will try to investigazte too.
Thanks

karawin, 2018-06-14:
Changed API to the new ones, but something else is wrong. Investigation again and again....

joaofl, 2018-06-17:
I saw some commits of yours. DO they solve the issue? Did you get ir working with IDF v3?

karawin, 2018-06-17:
No. I plan to integrate the old i2s driver in KaraDio32.
It seems a problem of bitrate but unable to find a solution.

joaofl, 2018-06-17:
Ok. It may be a temporary solution right? By replacing the deprecated I2S functions with the new implementations doesn't do the job?
I'll see what I can find as well.

karawin, 2018-06-17:
No, in fact in the code the old api is simply a call of the new api.
I will integrate the old driver and try to find where is the problem with the new code. May be i will ask some help on the espressif forum.

pilnikov, 2018-08-28:
lock my repo https://github.com/pilnikov/KaRadio32_fork. I updated and patched problem with i2s (Replacing with old). My ttgo play music fine.

pilnikov, 2018-09-12:
updated to 1.3r1. Does not work at full strength (screen problems) in version 2 bugs promised (JP) to eliminate

karawin, 2018-11-23:
I think we can close it now ;-)
https://youtu.be/Jk8d02b-uso

karawin, 2018-11-23:
http://karadio.karawin.fr/ttgo.zip
The spi init is in the vs1053. Yes some details must be finished.

pilnikov, 2018-11-26:
JP tnx, all work fine! Even with 255 for VS & display #199. Tnx!
Small bug - when switching stations, Cyrillic names are displayed in the wrong encoding for the first time.
1. Playing station #2 (Name on latin)
[image]
2. Tru change station on #0 (Name on cyrillic) - picture is not correct.
[image]
3. After changing the station, when the encoding also changed and changed the font
[image]
4. Try to switch the station again - everything shows correctly (font also changed)
[image]
[image]
If go back to step 1 - the bug will repeat.

karawin, 2018-12-16:
Must be ok now in 1.5 R5

pilnikov, 2018-12-17:
Without changes. Incorrect display of the title on the splash screen only.

karawin, 2018-12-18:
Yes sorry, now it's ok in 1.6 R0

karawin, 2018-12-19:
Great. I close it

closed 2018-12-19 by karawin