karawin/Ka-Radio32 #37: KaraDio32 with PCM5102 I2S DAC

Gromkz, 2018-06-27:
I use ESP32 with PCM5102 I2S DAC.
The project perfectly works with your bin files, too works with the OTA updates.

I want to add the IR codes.

But when I compile the project itself, there are problems with a sound, fast playback with noise and clicks. I have tried esp-idf versions from the last to esp-idf-v2.1. but I don't receive a correct sound.

Gromkz, 2018-06-28:
I try to enclose old i2s drivers.

karawin, 2018-06-28:
With the latest ESP-IDF you need to replace the i2s.c and .h given in the root of KaraDio32 in idf components driver

Gromkz, 2018-06-28:
And I have made.
the sound became better
but the sound strongly clicks at any updating of the screen
I have LCD_SPI_ILI9341

karawin, 2018-06-28:
I made some update recently for this problem

emax73, 2018-06-28:
Will update spi display speed from 1MHz to 20MHz, please -
its solve click problem.
Best Regards

Gromkz, 2018-06-28:
I update
git clone --recursive https://github.com/karawin/Ka-Radio32.git

replace the i2s.c and .h given in the root of KaraDio32 in idf components driver
clicks are after compilation
I update OTA of any clicks.

karawin, 2018-06-28:
I released a bin in R8 with all updates

emax73, 2018-06-28:
tested R8 - sound clicked with my SPI 256x64 display (SSD1322),
after increase SPI clock >= 10 MHz sound becomes continuous

karawin, 2018-06-28:
I lowered the spi speed because I was not sure that every type support the 10MHz.
May be i will make an uart command for that.

emax73, 2018-06-28:
u8g2 library has clock info for every controller, may using its

karawin, 2018-06-28:
You mean sck_pulse_width_ns?

For ucg
I use dev_config.clock_speed_hz   = (1000000000/((ucg_com_info_t *)data)->serial_clk_speed) ;

But not found where sck_pulse_width_ns is initialized.
Many comment like
* sck_pulse_width_ns = */  140,	
  /* sck_clock_hz = */ 1000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
  /* spi_mode = */ 0,		/* old: sck_takeover_edge, new: active high (bit 1), rising edge (bit 0) */
But ....

karawin, 2018-06-28:
May be
//		  dev_config.clock_speed_hz   = 1000000;
		  dev_config.clock_speed_hz   = u8x8->display_info->sck_clock_hz;
in u8g2_esp32_hal.c

emax73, 2018-06-29:
I see clock info at
u8x8_d_ssd1322.c, line 248
static const u8x8_display_info_t u8x8_ssd1322_256x64_display_info =
{
...
 /* sck_clock_hz = */ 10000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns, increased to 8MHz (issue 215), 10 MHz (issue 301) */
...
}

Gromkz, 2018-06-29:
today it was updated on OTA 
there were problems with a sound and on R8 (clicks) 

I have returned on R6

Gromkz, 2018-06-29:
I have compiled R7 with esp-idf-v2.1.1 
I have corrected main/websocket. with
Solves the issue of multiple declaration of the function 'base64_encode'

The sound good without clicks, IR works.
))

Gromkz, 2018-06-29:
R7 with the last esp-idf 
+ Solves the issue of multiple declaration of the function 'base64_encode'
+  old i2s drivers
works well

acolwill, 2018-08-24:
@Gromkz Any chance you could share your changes to the code to get the PCM5102 DAC working?
I've got a TTGO-TM-ESP32 board which has it integrated and am having a hard time getting it seen by Ka-Radio32.
Thanks in advance!

pilnikov, 2018-08-26:
https://github.com/pilnikov/KaRadio32_fork

acolwill, 2018-08-26:
@pilnikov Hi Alex, unfortunately I can't get that to work with my TTGO-TM-ESP32 unit... it will either not boot, or it won't initialise the screen

pilnikov, 2018-08-26:
If you compile with a new expressif sdk - you must comment line 72 in vs1053.c
	.flags = SPICOMMON_BUSFLAG_NATIVE_PINS|SPICOMMON_BUSFLAG_MASTER
but spi is not native

pilnikov, 2018-08-28:
I updated the files in my repository. You can try to collect. It works for me.

acolwill, 2018-08-28:
@pilnikov Thanks! Will have a look!

acolwill, 2018-08-28:
@pilnikov working brilliantly! Will communicate with you on your github page.