karawin/Ka-Radio32 #43: Heltec esp32

nano111, 2018-09-08:
Hi,
it is I again with my oled problems.
My oled diplay stoped working with no reason.I did not update anything.I have tried some other software for this boar and display dod not work.Now I ordered new oled 128x32 and connected it to gpio15 SCL and gpio4 as SDA.I cannot make it work.In gpio.h  pins are set correctly.What do I need to do when I chang something in gpio,h for the changes to take effect-only make flash???
.I have changed via serial sys.lcd to display type 6 but it does not work.Now I have version 1.3 R1.Here is serial outpt at start up:
E (4172) vs1053: VS1053/VS1003 detected. MP3Status: 40, Version: 4
device: 3ffd32a8
I2S Speed: 0
volume: 181
E (4532) main: LCD Type 6
X: 128, YY: 32, Y: 7
WIFI TRYING TO CONNECT TO SSID 1
E (5922) wifi: esp_wifi_disconnect 952 wifi not start
E (7182) main: 
Wifi Disconnected. Connection tried again
E (7182) main: 
Wifi Disconnected.
E (7312) main: 
Wifi Disconnected. Connection tried again
E (7312) main: 
Wifi Disconnected.
E (7442) main: 
Wifi Disconnected. Connection tried again
E (7442) main: 
Wifi Disconnected.
E (7582) main: 
Wifi Disconnected. Connection tried again
E (7582) main: 
Wifi Disconnected.
E (7912) event: invalid static ip
E (7912) main: 
Wifi connected

DNS: 192.168.0.1  
IP: 192.168.0.48

Init ............... Done
READY. Type help for a list of commands
autostart: playing:1, currentstation:6
##CLI.NAMESET#: 6 Naxie EX YU
##CLI.URLSET#: naxidigital-exyu128.streaming.rs
##CLI.PATHSET#: /listen.pls
##CLI.PORTSET#: 8240
##CLI.OVOLSET#: 0
##CLI.VOL#: 181
##CLI.URLSET#: naxidigital-exyu128.streaming.rs
##CLI.PATHSET#: /
##CLI.PORTSET#: 8240
##CLI.STOPPED# from C_PLIST
##CLI.ICY0#: NAXI EX YU RADIO (NAXI,Belgrade,Serbia, NAXI,Beograd,Srbija) - 128k
##CLI.ICY3#: http:\/\/www.naxi.rs
##CLI.ICY4#: EX YU Music
##CLI.ICY5#: 128
##CLI.PLAYING#
Radio works OK,only display not.Do I need to erase everything from board when make new flash.
How can I know which pins it actualy uses for display.What is been changed when I do OTA and disply used to stop working after that.

Thanks

karawin, 2018-09-12:
// I2C lcd (and rda5807 if lcd is i2c or LCD_NONE)
//------------------------------------------------
#define PIN_I2C_SCL GPIO_NUM_14
#define PIN_I2C_SDA GPIO_NUM_13
#define PIN_I2C_RST	GPIO_NUM_2		// or not used
Is the standard gpio.
Take care to change other that can be in conflict.
By default gpio4 is for the led.

nano111, 2018-09-12:
Thanks,
But when I do OTA, how to
get back to my own config of gpios.

Regards
Goran

karawin, 2018-09-12:
You cannot. Simply do a git pull, and generate again your bin.
gpio configuration is static.
May be in the future i will implement a local ota.

pilnikov, 2018-09-13:
maybe the best solution would be to configure gpio via serial as it is implemented for the led?

nano111, 2018-09-26:
Got new LCD,old one was broken.
Sorry for trouble

closed 2018-09-26 by nano111