karawin/Ka-Radio32 #44: Sound problem in new version 1.3R1

pilnikov, 2018-09-08:
if i use the version 1.3r1 with pcm5102 & ili9341 the sound is not clean. I solve the problem only by turning off the screen (set sys.lcd("0")). I think there is not enough memory at the same time to output to the screen and to decode the sound. on older versions of such a problem was not.

karawin, 2018-09-12:
Yes found it. A problem with spi when the output is i2s.
Corrected now in revision 2 available.

pilnikov, 2018-09-12:
Tnx. I check it ASAP

pilnikov, 2018-09-13:
I do not use the rotary encoder. In "stock" config sound is good, but system send warning about button3 (no encoder button)
[image]
[image]

if i comment encoder handler - the warning goes away, the picture on the screen changes, and I get a problem with the sound
[image]
[image]
[image]

karawin, 2018-09-13:
It is not a warning, it is a debug message.
Send
sys.logw
To the interface to change the log level to warning.

pilnikov, 2018-09-13:
[image]

no warning - sound bad

pilnikov, 2018-09-13:
I found the source of the problem - addonucg.c, function drawFrameUcg
[image]

karawin, 2018-09-13:
Try
void drawLinesUcg()
{
	setfont(text);
    for (int i=0;i<LINES;i++)
    {
		**taskYIELD();**
        if (mline[i]) draw(i); 
    }
}

pilnikov, 2018-09-13:
Also uncorrect drawing (on screen) symbols russian "ё" & "Ё" (code 1025 & 1105) - incomplete font conversion

karawin, 2018-09-13:
And a lot of other char. The used font is not complete
utf8To1251_t utf8To1251[UTF8TO1251] = {{0x401,0x45/*0xa8*/},{0x402,0x80},{0x403,0x81},{0x404,0xaa},{0x405,0xbd},{0x406,0x49/*0xb2*/},{0x407,0xaf},{0x408,0xa3},
									   {0x409,0x8a},{0x40a,0x8c},{0x40b,0x8e},{0x40c,0x8d},{0x40e,0xa1},{0x40f,0x8f},{0x452,0x90},{0x451,0x65/*0xb8*/},
									   {0x453,0x83},{0x454,0xba},{0x455,0xbe},{0x456,0x69/*0xb3*/},{0x457,0xbf},{0x458,0x6a/*0xbc*/},{0x459,0x9a},{0x45a,0x9c},
									   {0x45b,0x9e},{0x45c,0x9d},{0x45f,0x9f},{0x490,0xa5},{0x491,0xb4},
									   {0,0}};
uint8_t to1251(uint16_t utf8)
{
	int i;
	if (utf8 > 0x491) return 0x1f;
	for (i = 0; i<UTF8TO1251;i++)
	{
		if (utf8 == utf8To1251[i].utf8)
		{
//			printf("to1251: utf8: %x, ret: %x\n",utf8,utf8To1251[i].c1251);
			return utf8To1251[i].c1251;
		}
	}
	
//	printf("to1251: utf8: %x, ret: %x\n",utf8,(utf8 - 0x350)& 0xff);
	return ((utf8 - 0x350)& 0xff );
}

pilnikov, 2018-09-13:
i comment `ucg_DrawString(&ucg,0,yy - 30,0,strtime); 
in `void draw(int i)
time not drawing - sound ok, but ...
any dynamics on the screen (long titles - they need scrolling) - everything! the sound broke.

karawin, 2018-09-13:
Ok,
So in app_main.c line 981
replace
	xTaskCreatePinnedToCore (task_addon, "task_addon", 2600, NULL, 10, &pxCreatedTask,1);  //high priority for the spi else too slow due to ucglib
by
	xTaskCreatePinnedToCore (task_addon, "task_addon", 2600, NULL, 2, &pxCreatedTask,1);  

A test i forgot to remove....

pilnikov, 2018-09-14:
[image]
The sound is excellent only if the priority of the task "addons" is the lowest. The screen is very slow, updating approximately once every 4 seconds

karawin, 2018-09-14:
I don't understand why your config is so slow.
I tested a lcd 194 ili9341 with a dac output with and without a rotary, without problem with the last modification.
I cannot test with a PCM5102. I will order one to test it.

karawin, 2018-09-14:
Try level 4 please

karawin, 2018-09-15:
I have modified audio_renderer.c in order to improve the dma transfert.
Please take it (not yet finished but work well) at http://karadio.karawin.fr/test/audio_renderer.c
and put it in \esp\Ka-Radio32\components\audio_renderer
It may improve a lot the problem.

pilnikov, 2018-09-15:
The sound quality became much better (priority 4). Artifacts appear once every few(15-20) seconds. Sometimes floats the playback speed (not much) both in + and -.

karawin, 2018-09-16:
Take the new 1.3R3 release.

pilnikov, 2018-09-17:
close?

pilnikov, 2018-09-17:
upgrade to 1.3r3 - sound is agreeable

karawin, 2018-09-17:
ok, Good, i close it
Thanks.

closed 2018-09-17 by karawin