karawin/Ka-Radio32 #46: LCD ST7920 does not work correctly on 1.3

HIMjack666, 2018-09-09:
worked well on the version 0.9, but on 1.3 like this:
[image]

emax73, 2018-09-09:
SPI Clock for this controller is 1.25 MHz,
Now frequency 10 MHz,
maybe try

HIMjack666, 2018-09-09:
emax73, thank you a lot. Working good on 1 MHz.

closed 2018-09-09 by HIMjack666

ArnLil, 2021-05-09:
Hi, i have the same problem with ST7920, i am on the latest version 2.1.  For me this is a new install set-up. 
I also use a VS1053 for encoding the data.  When i choose DAC for example, the display is working fine. So there is a problem witch the combination of ST7920 and the VS1053.

Shtirlitsus, 2022-12-19:
same problem with combination of ST7920 and the VS1053. How to use different SPI interfaces on LCD and VS1053 decoder?

ArnLil, 2022-12-19:
HI, my problem was solved bij adding a diode in the data line.  (timing problem)

Shtirlitsus, 2022-12-19:
MOSI line?

ArnLil, 2022-12-19:
Not sure, but i think that is the data line.