karawin/Ka-Radio32 #50: **ERROR** multiple definition of `ucg_font_crox3c'

h1aji, 2018-09-18:
`/home/hmurad/esp/Ka-Radio32/build/ucglib/libucglib.a(ucg_karadio32_fonts.o):(.rodata.ucg_font_crox3c+0x0): multiple definition of `ucg_font_crox3c'
/home/hmurad/esp/Ka-Radio32/build/ucglib/libucglib.a(ucg_font_crox3c.o):(.rodata.ucg_font_crox3c+0x0): first defined here
collect2: error: ld returned 1 exit status
make: *** [/home/hmurad/esp/esp-idf/make/project.mk:411: /home/hmurad/esp/Ka-Radio32/build/KaRadio32.elf] Error `

I am trying to compile and seeing this error.
Is it possible to fix?

karawin, 2018-09-21:
make clean
delete ucg_font_crox3c.c
make app

h1aji, 2018-09-22:
which one should be deleted? 
```
./components/ucglib/csrc/ucg_font_crox3c.c
./components/ucglib/tools/font/karadio32/ucg_font_crox3c.c

```

karawin, 2018-09-22:
./components/ucglib/csrc/ucg_font_crox3c.c

closed 2018-09-22 by h1aji