karawin/Ka-Radio32 #51: WebServer fails on ESP32-Pico-D4

evilslon, 2018-09-20:
Hello! I have made my KaRadio32 using ESP32-Pico-D4 board taken from here => https://www.aliexpress.com/item/TTGO-Micro-32-Wifi-wireless-Bluetooth-Module-ESP32-PICO-D4-IPEX/32870840371.html 
I have built own firmware (to change some GPIOs), but there's one issue - KaRadio's internal webserver crashes after a couple of webpage updates, seems to be it is out of memory, but music is still playing.
UART output:
ets Jun  8 2016 00:22:57

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:2980
load:0x40078000,len:0
load:0x40078000,len:11400
entry 0x40078c38
I (91) main: starting app_main()
I (91) main: RAM left: 136052
E (91) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (131) main: Partition init done...
Log level is now ESP_LOG_ERROR
E (1741) vs1053: NO VS1053 detected
E (1741) main: LCD Type 0
X: 128, YY: 64, Y: 9
WIFI TRYING TO CONNECT TO SSID 1
E (3181) wifi: esp_wifi_disconnect 954 wifi not start
E (4431) event: invalid static ip
E (4431) main:
Wifi connected

DNS: 192.168.99.1
IP: 192.168.99.230

Init ............... Done
READY. Type help for a list of commands
autostart: playing:1, currentstation:249
##CLI.NAMESET#: 249 Супердискотека 90х
##CLI.URLSET#: online.radiorecord.ru
##CLI.PATHSET#: /sd90_128
##CLI.PORTSET#: 8102
##CLI.OVOLSET#: 0
##CLI.VOL#: 203
##CLI.ICY0#: \/sd90_128
##CLI.ICY3#: http:\/\/savonet.sf.net
##CLI.ICY4#: Misc
##CLI.ICY5#: 128
##CLI.ICY6#: Liquidsoap Radio!
##CLI.ICY7#:  channels=2;samplerate=44100;bitrate=128
##CLI.PLAYING#
##CLI.META#: YAKI-DA - I Saw You Dancing
E (16121) servers: Server low mem. Retrying...
E (18111) rmt: RMT[0] ERR
E (18111) rmt: status: 0x14000040
E (18801) servers: Server low mem. Retrying...
E (28691) servers: Server low mem. Retrying...
E (29091) servers: WebServer Socket fails accept errno: 23
E (30731) servers: WebServer Socket fails accept errno: 23
E (31101) servers: WebServer Socket fails accept errno: 23
E (31211) servers: WebServer Socket fails accept errno: 23
E (31321) servers: WebServer Socket fails accept errno: 23

sys.heap
##SYS.HEAP: 23156 #
```
After this, webserver is unavailable.

karawin, 2018-09-21:
Yes a bug in the esp-idf.
in esp\esp-idf\components\lwip\lwip\src\api\sockets.c
line 3545 change to 
int
lwip_close_r(int s)
{
  LWIP_API_LOCK();
#if ESP_LWIP
  LWIP_SET_CLOSE_FLAG(LWIP_SOCK_CLOSING);
#else
  LWIP_SET_CLOSE_FLAG();
#endif
  __ret = lwip_close(s);
#if ESP_LWIP
  if (__ret != 0  && EWOULDBLOCK == __sock->err)
  {
//  if (EWOULDBLOCK == __sock->err) {
    LWIP_SET_CLOSE_FLAG(LWIP_SOCK_OPEN);
  }
#endif
  LWIP_API_UNLOCK();
}

evilslon, 2018-09-21:
Thank you! Bug is gone with these changes! WebServer now works fine!

closed 2018-09-29 by karawin