karawin/Ka-Radio32 #52: Error when make flash

nano111, 2018-09-26:
I have downloaded new version of karadio32 from github,unziped it,made menuconfig and when  I do
make flash,I get this:

C:/msys32/home/l/ka34/main/vs1053.c: In function 'VS1053_spi_init':
C:/msys32/home/l/ka34/main/vs1053.c:75:25: error: 'KSPI' undeclared (first use i                                n this function)
  ret=spi_bus_initialize(KSPI, &buscfg, 1);  // dma
                         ^
C:/msys32/home/l/ka34/main/vs1053.c:75:25: note: each undeclared identifier is r                                eported only once for each function it appears in
In file included from C:/msys32/home/l/esp/esp-idf/components/esp32/include/esp_                                system.h:20:0,
                 from C:/msys32/home/l/ka34/main/include/vs1053.h:14,
                 from C:/msys32/home/l/ka34/main/vs1053.c:17:
C:/msys32/home/l/ka34/main/vs1053.c: In function 'VS1053_HW_init':
C:/msys32/home/l/ka34/main/vs1053.c:102:37: error: 'KSPI' undeclared (first use                                 in this function)
  ESP_ERROR_CHECK(spi_bus_add_device(KSPI, &devcfg, &vsspi));
                                     ^
C:/msys32/home/l/esp/esp-idf/components/esp32/include/esp_err.h:107:31: note: in                                 definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
make[1]: *** [/home/l/esp/esp-idf/make/component_wrapper.mk:286: vs1053.o] Error                                 1
make: *** [C:/msys32/home/l/esp/esp-idf/make/project.mk:481: component-main-buil                                d] Error 2

What is wrong?

karawin, 2018-09-28:
C:/msys32/home/l/ka34/main/vs1053.c:75:25: error: 'KSPI' undeclared (first use i n this function)
ret=spi_bus_initialize(KSPI, &buscfg, 1); // dma

Declared in ........\main\include\gpio.h
// Must be HSPI or VSPI
#define KSPI VSPI_HOST

Did you changed it?

nano111, 2018-09-29:
I put my own gpio.h because of the oled.
It was working in previous versions.All I had to do is replace original gpio.h with my gpio.h and make flash went OK.
Meanwhile , since I have new oled display , I connected it
like in original gpio.h so now I can do OTA update and it works.
Thanks
Regards
Goran

closed 2018-09-29 by karawin