karawin/Ka-Radio32 #55: Change gpio pins for ssd1306

luchino777, 2018-10-29:
Hi, I discovered that I2c pins on my esp32 wemos lolin, are different from the standard configuration of pins... is there a way to change them without compiling? Is it correct to changhe pins on the file gpio.h?
Thanks!
Luca

luchino777, 2018-11-02:
That's ok! changed in gpio.h, compiled and working... Only problem now is the lcd shows 1 pixel right, as reported here: [https://github.com/karawin/Ka-Radio32/issues/15](url) but that solution not working...

Luca

karawin, 2018-11-02:
The only difference between both lcd is this shift of one pixel, so the only solution is the command
sys.lcd("1")
and reset one time.

karawin, 2018-11-26:
No news? I will close it.
reopen if needed.

closed 2018-11-26 by karawin