karawin/Ka-Radio32 #57: ##CLI.STOPPED# from addon stop

luchino777, 2018-11-02:
Hi, compiled with gpio.h changed to get lcd working, on weemos lolin esp32.
It works.
Now gives me this error:  **##CLI.STOPPED# from addon stop** everytime it connects, it suddenly stops...

Which addon should I control that's giving this behaviour? Thanks

luchino777, 2018-11-02:
[https://www.hackster.io/johnnyfrx/esp32-with-integrated-oled-wemos-lolin-getting-started-07ac5d](url)
no cables.... it's a board with oled, esp32
Thanks

karawin, 2018-11-02:
By default the encoder uses
// Encoder knob
//-------------
#define PIN_ENC_A   GPIO_NUM_16		// CLK
#define PIN_ENC_B   GPIO_NUM_17		// DT
#define PIN_ENC_BTN GPIO_NUM_5		// SW

Try to wire the PIN 5 to vcc.

luchino777, 2018-11-05:
Hi, solved like this:
put #define PIN_ENCBTN_GPIO_NUM_15  //SW
and it's working.....
Only I do not know if 15 GPIO_NUM can be correct... 
I'm Newbie....

********************************************

// Encoder knob
//-------------
#define PIN_ENC_A   GPIO_NUM_16		// CLK
#define PIN_ENC_B   GPIO_NUM_17		// DT
#define PIN_ENC_BTN GPIO_NUM_15		// SW

// I2C lcd (and rda5807 if lcd is i2c or LCD_NONE)
//------------------------------------------------
#define PIN_I2C_SCL GPIO_NUM_4
#define PIN_I2C_SDA GPIO_NUM_5
#define PIN_I2C_RST	GPIO_NUM_2		// or not used

// I2C rda5807 (if lcd is spi)
// (removed)
//----------------------------
//#define PIN_SI2C_SCL GPIO_NUM_15
//#define PIN_SI2C_SDA GPIO_NUM_27

*****************************

Luca

karawin, 2018-11-05:
Correct, 15 or 27 or ... are free
I close this issue.
Enjoy

closed 2018-11-05 by karawin