karawin/Ka-Radio32 #59: esp_log.h : No such file or directory

shurikby, 2018-11-12:
Hi! When I try to compile version 1.4 and 1.5 i have this error:
/main/gpio.c:12:22: fatal error: esp_log.h : No such file or directory
Maybe you just forgot to add this file to the git repo?

h1aji, 2018-11-15:
I had same issue and there are couple of typos.
First in **main/gpio.c line 12 #include "esp_log.h "** space needs to be removed
Second in **main/CLickButtons.c** L should be lower case

@karawin can you please fix this?

karawin, 2018-11-15:
First in main/gpio.c line 12 #include "esp_log.h " space needs to be removed
Why not. no problem

Second in main/CLickButtons.c L should be lower case
Why?????

karawin, 2018-11-15:
Under mingw32 on windows10 no problem.
Is it a problem under  linux?

shurikby, 2018-11-15:
Yes, I have this problem under linux. But it seems that esp_log.h is just not presented in repo.

h1aji, 2018-11-16:
> Under mingw32 on windows10 no problem.
> Is it a problem under linux?

Linux is case-sensitive, so things like this will cause the error.

> Yes, I have this problem under linux. But it seems that esp_log.h just not presented in repo.
It does exist, there is a space symbol that causes this error.

h1aji, 2018-11-16:
@karawin Created two pull requests, please merge them. Should fix error on Linux platform. Thanks

karawin, 2018-11-16:
esp_log.h is an esp-idf include
https://github.com/espressif/esp-idf/blob/master/components/log/include/esp_log.h
I will remove the blanc char.

Uppercase are valid here.

Source has evolved. I don't merge. Done in next delivery,Thanks.

h1aji, 2018-11-16:
Can you please carefully look at this name "CLickButtons.h".

Especially second character. Capital L
This is causing issue in Linux

h1aji, 2018-11-16:
> CC build/main/ClickButtons.o
> /home/hmurad/esp/Ka-Radio32/main/ClickButtons.c:16:26: fatal error: ClickButtons.h: No such file or directory
> compilation terminated.
> make[1]: *** [/home/hmurad/esp/esp-idf/make/component_wrapper.mk:286: ClickButtons.o] Error 1
> make: *** [/home/hmurad/esp/esp-idf/make/project.mk:505: component-main-build] Error 2
>

karawin, 2018-11-16:
Yes, it was a L but i corrected it after the pre delivery.
Thanks

bazooka07, 2018-11-23:
I have Ubuntu 18.04 LTS and I have the same problem. Today, for compiling, I make
```
cd main/include
mv CLickButtons.h ClickButtons.h
cd ../../
make menuconfig
# update "Serial flasher config /  default serial port" to /dev/ttyUSB0
make
make flash
```
and it's works fine
Please, don't use uppercase letters in filenames for the main folder. To enhance some word, use underscore "_"
```
Don't use "<br/>" in README.md file. Markdown isn't HTML.
mv ClickButtons.h click_buttons.h
```

bazooka07, 2018-11-24:
Download the Zip archive of this repo and search in for  main/include/ClickButtons.h file
Not found ! In place, we have main/include/CLickButtons.h file
We have the same trouble by git clone ...

karawin, 2018-11-24:
OK, github on windows don't care about uppercase. 
Corrected now, sorry.

closed 2018-11-26 by karawin