karawin/Ka-Radio32 #65: Linker error

pilnikov, 2018-11-21:
When compiling, I get a link error. The error appeared after the IDF update. I think this is an IDF error, not a project.
[image]
Source is original. No changes were made.

pilnikov, 2018-11-21:
I spent 2 days trying to build version 1.5 with the updated ESP-IDF (https://github.com/espressif/esp-idf/releases/tag/v3.2-beta1) - I constantly received errors linker and esptool.py. Worked ONLY with the version of the ESP-IDF 3.1.1! Be careful!
JP please replace in nvs_partition_generator.sh -  ../esp-idf/ to "$ {IDF_PATH} /

closed 2018-11-21 by pilnikov

karawin, 2018-11-21:
#!/bin/bash
echo python ${IDF_PATH}/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py $1.csv build/$1.bin 8192
python ${IDF_PATH}/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py $1.csv build/$1.bin 8192
echo done

pilnikov, 2018-11-21:
ok! fine!

karawin, 2018-11-22:
#!/bin/bash
filename=$(basename "$1")
fname="${filename%.*}"
echo python ${IDF_PATH}/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py ${fname}.csv build/${fname}.bin 8192
python ${IDF_PATH}/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py ${fname}.csv build/${fname}.bin 8192
echo done

pilnikov, 2018-11-22:
very flexible - very good!