karawin/Ka-Radio32 #67: Ttgo VS1.4

nano111, 2018-11-24:
Hi,
I have bought this board https://github.com/LilyGO/TTGO-TS-V1.2,onlz VS1.4.
When I flash Karadio32  I get this :+1:

 rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6188
load:0x40078000,len:10168
load:0x40080400,len:6552
entry 0x40080764
[1B][0;32mI (56) boot: ESP-IDF v3.2-beta1-34-gfb7ba1baa 2nd stage bootloader[1B][0m
[1B][0;32mI (56) boot: compile time 15:22:43[1B][0m
[1B][0;32mI (65) boot: Enabling RNG early entropy source...[1B][0m
[1B][0;32mI (65) boot: SPI Speed      : 40MHz[1B][0m
[1B][0;32mI (67) boot: SPI Mode       : DIO[1B][0m
[1B][0;32mI (71) boot: SPI Flash Size : 4MB[1B][0m
[1B][0;32mI (75) boot: Partition Table:[1B][0m
[1B][0;32mI (79) boot: ## Label            Usage          Type ST Offset   Length[1B][0m
[1B][0;32mI (86) boot:  0 nvs              WiFi data        01 02 00009000 00006000[1B][0m
[1B][0;32mI (94) boot:  1 phy_init         RF data          01 01 0000f000 00001000[1B][0m
[1B][0;32mI (101) boot:  2 factory          factory app      00 00 00010000 00100000[1B][0m
[1B][0;32mI (109) boot: End of partition table[1B][0m
[1B][0;32mI (113) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x65320 (414496) map[1B][0m
[1B][0;32mI (267) esp_image: segment 1: paddr=0x00075348 vaddr=0x3ff80000 size=0x00000 (     0) load[1B][0m
[1B][0;32mI (267) esp_image: segment 2: paddr=0x00075350 vaddr=0x3ff80000 size=0x00000 (     0) load[1B][0m
[1B][0;32mI (274) esp_image: segment 3: paddr=0x00075358 vaddr=0x3ffb0000 size=0x0320c ( 12812) load[1B][0m
[1B][0;32mI (288) esp_image: segment 4: paddr=0x0007856c vaddr=0x3ffb320c size=0x00000 (     0) load[1B][0m
[1B][0;32mI (292) esp_image: segment 5: paddr=0x00078574 vaddr=0x40080000 size=0x00400 (  1024) load[1B][0m
[1B][0;32mI (301) esp_image: segment 6: paddr=0x0007897c vaddr=0x40080400 size=0x07694 ( 30356) load[1B][0m
[1B][0;32mI (322) esp_image: segment 7: paddr=0x00080018 vaddr=0x400d0018 size=0x9743c (619580) map[1B][0m
[1B][0;32mI (540) esp_image: segment 8: paddr=0x0011745c vaddr=0x40087a94 size=0x0cacc ( 51916) load[1B][0m
[1B][0;32mI (561) esp_image: segment 9: paddr=0x00123f30 vaddr=0x400c0000 size=0x00000 (     0) load[1B][0m
[1B][0;32mI (562) esp_image: segment 10: paddr=0x00123f38 vaddr=0x50000000 size=0x00000 (     0) load[1B][0m
[1B][0;32mI (568) esp_image: segment 11: paddr=0x00123f40 vaddr=0x50000000 size=0x00000 (     0) load[1B][0m
[1B][0;31mE (577) esp_image: Image length 1130352 doesn't fit in partition length 1048576[1B][0m
[1B][0;31mE (585) boot: Factory app partition is not bootable[1B][0m
[1B][0;31mE (591) boot: No bootable app partitions in the partition table[1B][0m
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

Gpio.h is changed for this board.

nano111, 2018-11-24:
After changing something in partiotions ...now I have this:
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6188
load:0x40078000,len:10168
load:0x40080400,len:6552
entry 0x40080764
[1B][0;32mI (29) boot: ESP-IDF v3.2-beta1-34-gfb7ba1baa 2nd stage bootloader[1B][0m
[1B][0;32mI (29) boot: compile time 17:54:41[1B][0m
[1B][0;32mI (38) boot: Enabling RNG early entropy source...[1B][0m
[1B][0;32mI (38) boot: SPI Speed      : 40MHz[1B][0m
[1B][0;32mI (39) boot: SPI Mode       : DIO[1B][0m
[1B][0;32mI (43) boot: SPI Flash Size : 4MB[1B][0m
[1B][0;32mI (47) boot: Partition Table:[1B][0m
[1B][0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[1B][0m
[1B][0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00004000[1B][0m
[1B][0;32mI (66) boot:  1 otadata          OTA data         01 00 0000d000 00002000[1B][0m
[1B][0;32mI (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000[1B][0m
[1B][0;32mI (81) boot:  3 ota_0            OTA app          00 10 00010000 001c0000[1B][0m
[1B][0;32mI (88) boot:  4 ota_1            OTA app          00 11 001d0000 001c0000[1B][0m
[1B][0;32mI (96) boot:  5 device           unknown          40 00 00390000 00001000[1B][0m
[1B][0;32mI (103) boot:  6 stations         unknown          41 00 00391000 00010000[1B][0m
[1B][0;32mI (111) boot:  7 device1          unknown          42 00 003a1000 00001000[1B][0m
[1B][0;32mI (118) boot:  8 hardware         WiFi data        01 02 003a2000 00002000[1B][0m
[1B][0;32mI (126) boot: End of partition table[1B][0m
[1B][0;32mI (130) boot: No factory image, trying OTA 0[1B][0m
[1B][0;32mI (135) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x65320 (414496) map[1B][0m
[1B][0;32mI (289) esp_image: segment 1: paddr=0x00075348 vaddr=0x3ff80000 size=0x00000 (     0) load[1B][0m
[1B][0;32mI (289) esp_image: segment 2: paddr=0x00075350 vaddr=0x3ff80000 size=0x00000 (     0) load[1B][0m
[1B][0;32mI (296) esp_image: segment 3: paddr=0x00075358 vaddr=0x3ffb0000 size=0x0320c ( 12812) load[1B][0m
[1B][0;32mI (310) esp_image: segment 4: paddr=0x0007856c vaddr=0x3ffb320c size=0x00000 (     0) load[1B][0m
[1B][0;32mI (314) esp_image: segment 5: paddr=0x00078574 vaddr=0x40080000 size=0x00400 (  1024) load[1B][0m
[1B][0;32mI (323) esp_image: segment 6: paddr=0x0007897c vaddr=0x40080400 size=0x07694 ( 30356) load[1B][0m
[1B][0;32mI (344) esp_image: segment 7: paddr=0x00080018 vaddr=0x400d0018 size=0x97438 (619576) map[1B][0m
[1B][0;32mI (561) esp_image: segment 8: paddr=0x00117458 vaddr=0x40087a94 size=0x0cacc ( 51916) load[1B][0m
[1B][0;32mI (583) esp_image: segment 9: paddr=0x00123f2c vaddr=0x400c0000 size=0x00000 (     0) load[1B][0m
[1B][0;32mI (584) esp_image: segment 10: paddr=0x00123f34 vaddr=0x50000000 size=0x00000 (     0) load[1B][0m
[1B][0;32mI (590) esp_image: segment 11: paddr=0x00123f3c vaddr=0x50000000 size=0x00000 (     0) load[1B][0m
[1B][0;32mI (611) boot: Loaded app from partition at offset 0x10000[1B][0m
[1B][0;32mI (611) boot: Disabling RNG early entropy source...[1B][0m
[1B][0;32mI (613) cpu_start: Pro cpu up.[1B][0m
[1B][0;32mI (617) cpu_start: Starting app cpu, entry point is 0x40081224[1B][0m
[1B][0;32mI (609) cpu_start: App cpu up.[1B][0m
[1B][0;32mI (627) heap_init: Initializing. RAM available for dynamic allocation:[1B][0m
[1B][0;32mI (634) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[1B][0m
[1B][0;32mI (640) heap_init: At 3FFCB4A0 len 00014B60 (82 KiB): DRAM[1B][0m
[1B][0;32mI (646) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[1B][0m
[1B][0;32mI (653) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[1B][0m
[1B][0;32mI (659) heap_init: At 40094560 len 0000BAA0 (46 KiB): IRAM[1B][0m
[1B][0;32mI (665) cpu_start: Pro cpu start user code[1B][0m
[1B][0;32mI (12) cpu_start: Starting scheduler on PRO CPU.[1B][0m
[1B][0;32mI (0) cpu_start: Starting scheduler on APP CPU.[1B][0m
[1B][0;32mI (14) main: starting app_main()[1B][0m
[1B][0;32mI (14) main: RAM left: 198736[1B][0m
[1B][0;31mE (14) main: Running partition type 0 subtype 16 (offset 0x00010000)[1B][0m
[1B][0;32mI (74) main: Partition init done...[1B][0m
Log level is now ESP_LOG_ERROR
[1B][0;31mE (214) spi: spicommon_bus_initialize_io(188): not using iomux pins[1B][0m
assertion "ret==ESP_OK" failed: file "C:/msys32/home/l/esp/ka32/main/vs1053.c", line 88, function: VS1053_spi_init
abort() was called at PC 0x400e8beb on core 0

Backtrace: 0x40091c1c:0x3ffcd140 0x40091e49:0x3ffcd160 0x400e8beb:0x3ffcd180 0x400db8c9:0x3ffcd1b0 0x400d59d0:0x3ffcd200 0x400d2df2:0x3ffcd240 0x4008facd:0x3ffcd260

Rebooting...

karawin, 2018-11-25:
Cannot answer. I have no idea of what you flashed and where.
Do you generate your KaRadio32.bin?
Do you generated an hardware partition bin or  modified the gpio.h directly etc etc.

try the config i made for a ttgo TM. It is very close apart the buttons gpio.
[ttgo.zip](https://github.com/karawin/Ka-Radio32/files/2612960/ttgo.zip)

nano111, 2018-11-25:
Thanks
I flash vi esp-idf and mingw32 via command make flash.Before I edit gpio.h file.
I know you do not have time ,and it it is easy for you to do this stuff,but for as newbies it is hard.
Please how to use that config you made for ttgo --via mingw32 or I have to use loader tool?
Thanks for your time

karawin, 2018-11-25:
Do not modify gpio.h. Restore it.
Restore the bin to the default with 
> make app.
Adapt ttgo.csv to your needs, 
Call
 > ./nvs_partition_generator.sh ttgo.csv

And flash all with (adapt yourhome and com)
python ${IDF_PATH}/components/esptool_py/esptool/esptool.py \
   --chip esp32 --port com5 --baud 460800 --before default_reset \
   --after hard_reset write_flash -u --flash_mode dio \
   --flash_freq 40m --flash_size detect \
   0x1000 /home/yourhome/esp/Ka-Radio32/build/bootloader/bootloader.bin \
   0x10000 /home/yourhome/esp/Ka-Radio32/build/KaRadio32.bin \
   0x8000 /home/yourhome/esp/Ka-Radio32/build/partitions.bin \
   0x3a2000 /home/yourhome/esp/Ka-Radio32/build/ttgo.bin

nano111, 2018-11-25:
When I try to generate bin ....this 
$ ./nvs_partition_generator.sh ttgo.csv
python C:/msys32/home/l/esp/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py ttgo.csv build/ttgo.bin 8
Size of partition (must be multiple of 4096)
done

and no ttgo.bin anywhere.
I have tried with yours ttgo.bin...radio starts but no display becaus of wrong gpio.

karawin, 2018-11-25:
try with 8192 in place of 8 or 0x2000
I know, it is curious, but it seems they changed without notification.

nano111, 2018-11-25:
Now I used old_nvs_partition_generator-sh and it built ttgo.bin.
After that sam rebooting as befor....have to find right gpio

nano111, 2018-11-25:
still no display.Changed gpio,radio starts normaly ...but when I change to sys.lcd("192")
which is my display according to seller st7735 after reboot it starts to rebbot constantly....

Log level is now ESP_LOG_ERROR
[1B][0;31mE (314) vs1053: VS1053 not used[1B][0m
[1B][0;31mE (314) main: LCD Type 192[1B][0m
[1B][0;31mE (404) spi_master: spi_bus_add_device(427): invalid host[1B][0m
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4009205c
file: "C:/msys32/home/l/esp/ka32/components/ucglib/csrc/ucg_esp32_hal.c" line 165
func: ucg_com_hal
expression: spi_bus_add_device(ucg_esp32_hal.spi_no, &dev_config, &handle)

Backtrace: 0x40091c1c:0x3ffccfe0 0x4009205f:0x3ffcd000 0x401026b9:0x3ffcd020 0x4010430d:0x3ffcd090 0x401030de:0x3ffcd0c0 0x40103de1:0x3ffcd0f0 0x40103aed:0x3ffcd110 0x40103194:0x3ffcd130 0x400e318f:0x3ffcd150 0x400d8e39:0x3ffcd1e0 0x400d5a0c:0x3ffcd200 0x400d2df2:0x3ffcd240 0x4008facd:0x3ffcd260

Rebooting...

karawin, 2018-11-25:
Send me your csv
It is the spi which is incorrect.

karawin, 2018-11-25:
Something like that
[ttgov2.zip](https://github.com/karawin/Ka-Radio32/files/2613266/ttgov2.zip)

nano111, 2018-11-25:
My csv:
key,type,encoding,value
gpio_space,namespace,,
K_SPI,data,u8,255
P_MISO,data,u8,2
P_MOSI,data,u8,23
P_CLK,data,u8,5
P_XCS,data,u8,255
P_RST,data,u8,255
P_XDCS,data,u8,255
P_DREQ,data,u8,255
P_ENC0_A,data,u8,255
P_ENC0_B,data,u8,38
P_ENC0_BTN,data,u8,255
P_ENC1_A,data,u8,255
P_ENC1_B,data,u8,17
P_ENC1_BTN,data,u8,255
P_BTN0_A,data,u8,34
P_BTN0_B,data,u8,35
P_BTN0_C,data,u8,39
P_BTN1_A,data,u8,255
P_BTN1_B,data,u8,255
P_BTN1_C,data,u8,255
P_I2C_SCL,data,u8,26
P_I2C_SDA,data,u8,27
P_I2C_RST,data,u8,32
P_LCD_CS,data,u8,16
P_LCD_A0,data,u8,17
P_LCD_RST,data,u8,9
P_IR_SIGNAL,data,u8,255
P_I2S_LRCK,data,u8,25
P_I2S_BCLK,data,u8,26
P_I2S_DATA,data,u8,19
P_ADC,data,u8,255
custom_ir_space,namespace,,
K_UP,data,string,
K_LEFT,data,string,
K_OK,data,string,
K_RIGHT,data,string,
K_DOWN,data,string,
K_0,data,string,
K_1,data,string,
K_2,data,string,
K_3,data,string,
K_4,data,string,
K_5,data,string,
K_6,data,string,
K_7,data,string,
K_8,data,string,
K_9,data,string,
K_STAR,data,string,
K_DIESE,data,string,
K_INFO,data,string,

karawin, 2018-11-25:
Please read https://github.com/karawin/Ka-Radio32/blob/master/HardwareConfig.md
key,type,encoding,value
gpio_space,namespace,,
K_SPI,data,u8,2
P_MISO,data,u8,19
P_MOSI,data,u8,23
P_CLK,data,u8,18
is mandatory for the spi bus.

nano111, 2018-11-25:
[image]
 this is my gpio i new csv
key,type,encoding,value
gpio_space,namespace,,
K_SPI,data,u8,2
P_MISO,data,u8,19
P_MOSI,data,u8,23
P_CLK,data,u8,5
P_XCS,data,u8,255
P_RST,data,u8,33
P_XDCS,data,u8,2
P_DREQ,data,u8,14
P_ENC0_A,data,u8,255
P_ENC0_B,data,u8,38
P_ENC0_BTN,data,u8,255
P_ENC1_A,data,u8,255
P_ENC1_B,data,u8,17
P_ENC1_BTN,data,u8,255
P_BTN0_A,data,u8,34
P_BTN0_B,data,u8,35
P_BTN0_C,data,u8,39
P_BTN1_A,data,u8,255
P_BTN1_B,data,u8,255
P_BTN1_C,data,u8,255
P_I2C_SCL,data,u8,5
P_I2C_SDA,data,u8,16
P_I2C_RST,data,u8,17
P_LCD_CS,data,u8,16
P_LCD_A0,data,u8,17
P_LCD_RST,data,u8,9
P_IR_SIGNAL,data,u8,255
P_I2S_LRCK,data,u8,25
P_I2S_BCLK,data,u8,26
P_I2S_DATA,data,u8,19
P_ADC,data,u8,255
custom_ir_space,namespace,,
K_UP,data,string,
K_LEFT,data,string,
K_OK,data,string,
K_RIGHT,data,string,
K_DOWN,data,string,
K_0,data,string,
K_1,data,string,
K_2,data,string,
K_3,data,string,
K_4,data,string,
K_5,data,string,
K_6,data,string,
K_7,data,string,
K_8,data,string,
K_9,data,string,
K_STAR,data,string,
K_DIESE,data,string,
K_INFO,data,string,

karawin, 2018-11-25:
No
P_CLK,data,u8,5 is wrong

P_CLK,data,u8,18

Try the one i sent.

karawin, 2018-11-25:
Because P_CLK must be
P_CLK,data,u8,18
The soft for now is waiting native gpio for the spi.
To change that the line 84 in vs1053.c
.flags = SPICOMMON_BUSFLAG_NATIVE_PINS|SPICOMMON_BUSFLAG_MASTER
must be changed to 
.flags = SPICOMMON_BUSFLAG_MASTER

I suppose but I can't help you if i don't see the log.

nano111, 2018-11-26:
Thanks ....that did the trick.
Made change to vs1053.c, then make app ...then changed P_CLK to gpio 5 and it started.
I have one raw of pixels on the top  and right side that are garbled...but it works.
now if I want to attach VS1053 board....will it work? 
Thanks again....learning slowly with your help   :-)

karawin, 2018-11-26:
Good news
I changed too in the last release 1.5R2, so now you can do an OTA without having to regenerate the bin yourself.
The vs1053 will work if you have enough free gpio and configure it in your csv.
No problem.

nano111, 2018-11-26:
Just tried OTA...works.
Can you solve this shift on the display=

[image]

nano111, 2018-11-26:
sorry ...picture is rotated
[image]

nano111, 2018-11-26:
Alredy tried...the best is 192 as on picture

karawin, 2018-11-26:
In the log after reset?
E (492) main: LCD Type 192
X: 160, yy: 128, y: 14

I will try to see if another type exists in ucglib.

nano111, 2018-11-26:
yes

nano111, 2018-11-26:
Maybe this helps....from seller specs of display:

SST7735 | Description
-- | --
Size | 0.96/1.44" diagonal LCD TFT display
Resolution | 128x128/160x80 resolution,  16-bit/18-bit (262,144) color
Voltage | Onboard 3.3V @ 150mA LDO regulator
Bus | 5 wire SPI digital interface:CS:16, A0:17, SDA:23, SCK:5, RST:9
Dissipation | Current draw is based on LED backlight usage: with full backlight draw is ~25mA

karawin, 2018-11-26:
It seems to be a 128x128. That's why...
I try to adapt with a new type.

nano111, 2018-11-26:
OK thanks.
One more beer for you..... :-)

karawin, 2018-11-26:
I like beer ;-)

karawin, 2018-11-26:
tried 193 and 195?

karawin, 2018-11-26:
OTA and 
sys.lcd("200")

nano111, 2018-11-26:
Tried 193 and 195
this is with 200

[image]

karawin, 2018-11-26:
I think it will be bad. It is really a 128x160

nano111, 2018-11-26:
Not OK,same as before

nano111, 2018-11-26:
Maybe if you can put spaces in the lines before station name and meta data it will be good enough ,just to see the whole first letters.

nano111, 2018-11-26:
so the text is shifted one space to the right

nano111, 2018-11-26:
or to put it in the center of display..I do not know how dificult it is

karawin, 2018-11-26:
No, the code is the same for all king of lcd.
I am using this kind of lcd without problem 192 and 194
[image]

nano111, 2018-11-26:
this one is embedded with  the esp32 board.Maybe olikraus will know something more.Buttons work OK.
Thanks for now.If you here something new,please let me know.
Good night

karawin, 2018-11-26:
May be a 160x80. So i will try it tomorrow.
Good night

karawin, 2018-11-27:
You can try lcd 201  (80x160) but i don't think it is the good one. 
It seems a real 128x160 but with a shift.
Can i have a better snap? I see some strange pixels.

nano111, 2018-11-27:
With 201 screen overlaps...no good, this is with 192
[image]

karawin, 2018-11-27:
Ok,  After scanning the web, i understand that some st7735 range to 2- 129 and other 0- 127
That explain the 2 pixels on to and right.

So i will create another 128x160 type with shift.

nano111, 2018-11-27:
ok

karawin, 2018-11-28:
See https://github.com/olikraus/ucglib/issues/106#issuecomment-442222681

Please OTA and lcd 202
I made a 132x162 to see what happen.

nano111, 2018-11-28:
Much better.....only first raw of meta data is not visible....
[image]

nano111, 2018-11-28:
When I rotate screen it is the same.First pixsels are not shown.

nano111, 2018-12-01:
No....didyou mean lcd 203?
strange pxels at teh bottom...clock and date almost not visible.
Last one was better...just to instert one space on the left size of the screen
[image]

karawin, 2018-12-02:
new test please

nano111, 2018-12-02:
Even more shifted to the left
[image]

karawin, 2018-12-10:
Hi

A new test with 202 please.

nano111, 2018-12-11:
Same as on the picture above.

karawin, 2018-12-11:
same problem on the blue bar of the volume?
And with rotation?
It seems that it is not a 160x230

nano111, 2018-12-12:
Same with rotation as on picture above the last.
Didn't notice blue  bar,I will send you the picture later.
Thanks for your effort

karawin, 2018-12-12:
Sorry, i made a lot of mistake in the ota. Please try again and thanks to be patient.

nano111, 2018-12-12:
Perfect!
Thank YOU for your patient.
Another beer for you!
[image]

karawin, 2018-12-12:
Great news.
I wonder if the rotation is working?

Anyway it is usable as it is now.
Pfiouu.

nano111, 2018-12-12:
Rotation works.

closed 2018-12-13 by nano111