karawin/Ka-Radio32 #68: How to install the Ka-Radio32 to ESP32 board?

wieb18, 2018-11-24:
Hello,

I am very very new to Arduino, just few days ago bought new ESP32 and VS1053 boards.

[image]

[image]

Trying flash Ka-Radio32 to my ESP32 with ESP Flash Download Tool v2.3:

[image]

And... the result is **FAIL**

[image]

The error message: **CHIP SYNC ERROR**

But when I press BOOT button on ESP32 then click START button on ESP Flash Download Tool v2.3:

[image]

Chip Sync Ok but different error message: **READ REG ERROR**

Again... flashing my ESP32 is FAILED.

[image]

Please help, **how to install the Ka-Radio32 to ESP32 board?**

Any help very appreciated and thanks in advance.

nano111, 2018-11-24:
Maybe you have wrong flash size,try 4Mbit.Try esp flash tool v3.6.2,your is 2.3

karawin, 2018-11-25:
Yes, try a v3.6 tool.
I never had to press boot, but if you want to use it, press and hold Boot, the press Rst and release boot.

wieb18, 2018-11-25:
> Maybe you have wrong flash size,try 4Mbit.Try esp flash tool v3.6.2,your is 2.3

Thank you for the guidance.

Using Flash Download Tool v3.6.5 download from [here](https://www.espressif.com/en/support/download/other-tools?keys=&field_type_tid%5B%5D=13)

[image]

Now I am able flashing ESP32 without any issue and without pressing any button.

But what I should do next?

Scanning wifi network, WifiKaRadio SSID is not found. 

ESP32 Serial output:

[image]

Please help....

nano111, 2018-11-25:
Try to erase flash first.It worked for me once.

wieb18, 2018-11-25:
> When i see the serial, the r device is not flashed.
> Power off the esp32.
> Power on.
> Flash it with a higher speed 921600 or up.
> When finish on the tool, launch the serial
> Reboot the esp32

[image]

[image]

**Can't go higher than 921600**

[image]

[image]

**Max speed is 921600**

> Try to erase flash first.It worked for me once.

Here is what just I did:
1. Erase flash
2. Power off then power on
3. Flashing with ESP Download Tool v3.6.5. (speed 921600)
4. Power off then power on

Then... **nothing happen**

Nothing on Serial Monitor and still no WifiKaRadio SSID.

Also nothing happen on Serial Monitor when press BOOT button.

But when I press RESET button, the Serial Monitor showing this:

[image]

Don't know what it means. Please help....

karawin, 2018-11-25:
So, it started but crashed.
the KaRadio32.bin is the default one or do you changed something in the gpio.h before generating it?
Many errors come from a wrong configuration.
Do you wired the vs1053?
If yes, it may be not well wired. Try without it.

A good start is
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3580
load:0x40078000,len:9300
load:0x40080400,len:5764
entry 0x400806d4
I (119) main: starting app_main()
I (119) main: RAM left: 160852
E (119) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (159) main: Partition init done...
Log level is now ESP_LOG_ERROR
E (4249) vs1053: VS1053/VS1003 detected. MP3Status: 40, Version: 4
device: 3ffcd384
I2S Speed: 0
volume: 112
E (4569) main: LCD Type 194
X: 320, yy: 240, y: 23
WIFI TRYING TO CONNECT TO SSID 1
E (5519) wifi: esp_wifi_disconnect 954 wifi not start
E (7169) event: invalid static ip
E (7169) main: 
Wifi connected

wieb18, 2018-11-25:
> So, it started but crashed.
> the KaRadio32.bin is the default one or do you changed something in the gpio.h before generating it?
>
I am directly flashing Ka-Radio to ESP32 with ready made BIN files, so I don't have any chance to change anything. Also I am very very new to Arduino (just a few days), still don't have any clue to change anything.

> Do you wired the vs1053?
> If yes, it may be not well wired. Try without it.
> 

You are correct, the ESP32 wired to VS1053. Although VS1053 in power off condition but because the wires still attached between VS1053 and ESP32 making Ka-Radio crash. After removing all the wires, Ka-Radio finally running.

[image]

Serial Monitor output:

[image]

And here is my wiring between ESP32 and VS1053:

[image]

**My questions:**
1. Is my wiring above already correct?
2. How to connect Ka-Radio to my Wifi network? Where I put SSID and password of my own Wifi?

Thanks a lot for the help, very appreciated.

karawin, 2018-11-25:
No, your scheme is for another project under arduino.
Here:
#define PIN_NUM_XCS  GPIO_NUM_32
#define PIN_NUM_RST  GPIO_NUM_12 
#define PIN_NUM_XDCS GPIO_NUM_33
#define PIN_NUM_DREQ GPIO_NUM_34

That explain the problem.
Your web page is not loaded. correctly.

wieb18, 2018-11-25:
> Your web page is not loaded. correctly.

Yes, on the first run the web page took forever to load, its very slow. But after power off then power on the ESP32, it load fast (without any wires attached between ESP32 &  VS1053).

[image]

Successfully changed my wifi preference on Setting Web Page.

[image]

> No, your scheme is for another project under arduino.

Thanks for the guidance, updated my wiring into this:
1. SCK to GPIO 18
2. MISO to GPIO 19
3. MOSI to GPIO 23
4. XRESET to GPIO 12
5. XDCS to GPIO 33
6. XCS to GPIO 32
7. DREG to GPIO 34

Then power off and on the ESP32 also power on VS1053.

But the result, Ka-Radio web page not able to load.

[image]

Pressing RESET button on ESP32, the Serial Monitor displaying this message:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3580
load:0x40078000,len:9300
load:0x40080400,len:5764
entry 0x400806d4
I (118) main: starting app_main()
I (118) main: RAM left: 160852
E (118) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (158) main: Partition init done...
Log level is now ESP_LOG_ERROR
E (4188) vs1053: VS1053/VS1003 detected. MP3Status: 0, Version: 0
device: 3ffcd384
volume: 162
E (4288) main: LCD Type 0
X: 128, YY: 64, Y: 9
WIFI TRYING TO CONNECT TO SSID 1
E (5778) wifi: esp_wifi_disconnect 954 wifi not start
E (7128) event: invalid static ip
E (7128) main: 
Wifi connected

DNS: 192.168.0.230  
IP: 192.168.0.240

gpio_get_button0 error 4354
gpio_get_button1 error 4354
gpio_get_adc error 4354
E (13558) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (13558) task_wdt:  - IDLE0 (CPU 0)
E (13558) task_wdt: Tasks currently running:
E (13558) task_wdt: CPU 0: task_lcd
E (13558) task_wdt: CPU 1: IDLE1
E (18548) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (18548) task_wdt:  - IDLE0 (CPU 0)
E (18548) task_wdt: Tasks currently running:
E (18548) task_wdt: CPU 0: esp_timer
E (18548) task_wdt: CPU 1: IDLE1

```
As could see on Serial Monitor, now **VS1053/VS1003 detected**.

But why Ka-Radio web page very slow and I am not hear any sound from VS1053?
Is my wiring wrong again? 

Please help....

karawin, 2018-11-25:
Something wrong in the vs1053

Detected because DREQ line comes low. Ok, but the registers are not read.
Check again.
And this is blocking so other symptoms.
Other strange thing
gpio_get_button0 error 4354
gpio_get_button1 error 4354
gpio_get_adc error 4354
It seems that the standard_adb.bin is not up to date.

[standard_adb.zip](https://github.com/karawin/Ka-Radio32/files/2613450/standard_adb.zip)

wieb18, 2018-11-25:
>
> It seems that the standard_adb.bin is not up to date.
>

[image]

Strange... I got the standard_adb.bin and all the KA-Radio BIN files from your Github page. Maybe you also need updating the standard_adb.bin on KaRadio32sup.zip file to prevent anyone has the same issue like me.

Once again I do erasing and flashing my ESP32 with the new standard_adb.bin including all the BIN files then double check my wiring between ESP32 and VS1053.

[image]

Finally... VS1053 detected by KA-Radio. But why the **volume has negative value**?

Serial Monitor output:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3580
load:0x40078000,len:9300
load:0x40080400,len:5764
entry 0x400806d4
I (118) main: starting app_main()
I (118) main: RAM left: 160852
E (118) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (158) main: Partition init done...
Log level is now ESP_LOG_ERROR
E (4198) vs1053: VS1053/VS1003 detected. MP3Status: 40, Version: 4
device: 3ffcd384
I2S Speed: 0
volume: 158
E (4518) main: LCD Type 0
X: 128, YY: 64, Y: 9
WIFI TRYING TO CONNECT TO SSID 1
E (6008) wifi: esp_wifi_disconnect 954 wifi not start
E (7338) event: invalid static ip
E (7338) main: 
Wifi connected

DNS: 192.168.0.230  
IP: 192.168.0.240

Init .......... Done
READY. Type help for a list of commands
##CLI.NAMESET#: 0 Europe1
##CLI.URLSET#: e1-live-mp3-128.scdn.arkena.com
##CLI.PATHSET#: /europe1.mp3
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 14
##CLI.VOL#: 158
##CLI.PLAYING#

##CLI.NAMESET#: 10 FIP Rock
##CLI.URLSET#: www.listenlive.eu
##CLI.PATHSET#: /fr_fiprock.m3u
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 158
##CLI.URLSET#: direct.fipradio.fr
##CLI.PATHSET#: /live/fip-webradio1.mp3?ID=f9fbk29m84
##CLI.PORTSET#: 80
##CLI.STOPPED# from C_PLIST
##CLI.STOPPED# from C_HDER
##CLI.URLSET#: chai5she.cdn.dvmr.fr
##CLI.PATHSET#: /fip-webradio1.mp3?ID=f9fbk29m84
##CLI.PORTSET#: 80
##CLI.PLAYING#

##CLI.NAMESET#: 10 FIP Rock
##CLI.URLSET#: www.listenlive.eu
##CLI.PATHSET#: /fr_fiprock.m3u
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 158
##CLI.URLSET#: direct.fipradio.fr
##CLI.PATHSET#: /live/fip-webradio1.mp3?ID=f9fbk29m84
##CLI.PORTSET#: 80
##CLI.STOPPED# from C_PLIST
##CLI.STOPPED# from C_HDER
##CLI.URLSET#: chai5she.cdn.dvmr.fr
##CLI.PATHSET#: /fip-webradio1.mp3?ID=f9fbk29m84
##CLI.PORTSET#: 80
##CLI.PLAYING#

##CLI.NAMESET#: 34 MusicOnly FM
##CLI.URLSET#: listen.radionomy.com
##CLI.PATHSET#: /musiconly-fm
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 158
##CLI.STOPPED# from C_HDER
##CLI.URLSET#: streaming.radionomy.com
##CLI.PATHSET#: /MusicOnly-FM
##CLI.PORTSET#: 80
E (126848) webclient: Client socket: 63  read: -1  errno:11 
##CLI.STOPPED# from No Data
##CLI.ICY0#: No Data
##CLI.NAMESET#: 30 Nostalgie
##CLI.URLSET#: www.listenlive.eu
##CLI.PATHSET#: /fr_nostalgie.m3u
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 158
##CLI.URLSET#: cdn.nrjaudio.fm
##CLI.PATHSET#: /audio1/fr/30601/mp3_128.mp3?origine=listenlive
##CLI.PORTSET#: 80
##CLI.STOPPED# from C_PLIST
##CLI.STOPPED# from C_HDER
##CLI.URLSET#: 185.52.127.160
##CLI.PATHSET#: /fr/30601/mp3_128.mp3?origine=listenlive
##CLI.PORTSET#: 80
##CLI.PLAYING#

CORRUPT HEAP: multi_heap.c:429 detected at 0x3ffeda78
abort() was called at PC 0x400907ff on core 0

Backtrace: 0x40090e9e:0x3ffd6d70 0x400910b9:0x3ffd6d90 0x400907ff:0x3ffd6db0 0x40082ac5:0x3ffd6dd0 0x40082af1:0x3ffd6df0 0x40082efd:0x3ffd6e10 0x400830c9:0x3ffd6e50 0x400865ad:0x3ffd6e70 0x4012b607:0x3ffd6e90 0x401356c1:0x3ffd6ed0 0x40089929:0x3ffd6f50 0x4008ed5d:0x3ffd6f90

Entering gdb stub now.
$T0b#e6

```

The radio sound is very choppy and after a few minutes KA-Radio crash.

Please guide me... where I done wrong?

wieb18, 2018-11-25:
After the crashed above, I pressed RESET button and here is the Serial Monitor output:

[image]

The ESP32 not run Wifi and Ka-Radio but directly entering Task Watchdog.

Pressing again RESET button and here is the result:

[image]

Now ESP32 connected to Wifi and able to run Ka-Radio (with choppy sound). But only for a few second and then crash again entering Task WatchDog loop.

What is Task WatchDog?

karawin, 2018-11-25:
Volume is negative because it is an attenuation. Full volume is 0 decibel

How do you connected the power lines and ground?

karawin, 2018-11-25:
There eratic logs are typical of a power problem.

wieb18, 2018-11-25:
> 
> Volume is negative because it is an attenuation. Full volume is 0 decibel
> 

Thank you for the explanation.

> 
> How do you connected the power lines and ground?
>

The ESP32 power by 5V micro USB from PC and VS1053 power by 5V separate power supply 2Amp

karawin, 2018-11-25:
And ground of each connected together?

karawin, 2018-11-25:
I use to connect the 5v of the vs1053 board with the 5V of the esp32, and gnd obviouslly. No external power supply.

wieb18, 2018-11-25:
> And ground of each connected together?

No....

I am following the wiring according Ka-Radio Github page:

```
// KSPI pins of the SPI bus
//-------------------------
#define PIN_NUM_MISO GPIO_NUM_19 	// Master Input, Slave Output
#define PIN_NUM_MOSI GPIO_NUM_23	// Master Output, Slave Input   Named Data or SDA or D1 for oled
#define PIN_NUM_CLK  GPIO_NUM_18 	// Master clock  Named SCL or SCK or D0 for oled

// gpio of the vs1053
//-------------------
#define PIN_NUM_XCS  GPIO_NUM_32
#define PIN_NUM_RST  GPIO_NUM_12 
#define PIN_NUM_XDCS GPIO_NUM_33
#define PIN_NUM_DREQ GPIO_NUM_34
// + KSPI pins

```

wieb18, 2018-11-25:
> I use to connect the 5v of the vs1053 board with the 5V of the esp32, and gnd obviouslly. No external power supply.

I will try your guidance ASAP, hope it will solve the issue

karawin, 2018-11-25:
It will.Every electronic equipment need a common gnd (apart audio amplifier but it is a particular case)

wieb18, 2018-11-26:
Tried using only 1 power source for both ESP32 & VS1053:
1. Using only 5V micro USB powering from PC
2. Using only 5V 2Amp external power source 

Both result Ka-Radio works well (no crash as previously) and the sound also very good, no choppy sound anymore. 

Thanks a lot for guiding me all this time, you are the best

karawin, 2018-11-26:
As i told you before, it is NOT an arduino project. It uses directly the esp-idf.
Start in the main directory app_main.c
The code is a multi tasks under freertos.
Sorry, no init, no loop ;-)

I am happy to see that all is resolved.
I will close the issue because it is solved.
Bye

closed 2018-11-26 by karawin