karawin/Ka-Radio32 #79: Sys.erase not clean a playlist

pilnikov, 2018-12-23:
If I use sys.erase my playlist is not cleared completely, but is damaged (if use it - invalid host). It would be nice if the playlist was cleared completely.

shurikby, 2018-12-27:
Press "Refresh local list" button. Playlist is clear I guess, you just see cached page.

karawin, 2018-12-30:
sys.erase clean all configuration (wifi etc) and stations (playlist).
It is not a good command to erase only  the stations list.

Noted in https://github.com/karawin/Ka-Radio32/blob/master/Interface.md
sys.erase		: erase all recorded configuration and stations.

To erase the list only on the web page "edit" use the Erase all button.

But it is a good remark. I will create a serial command for that.

closed 2019-01-03 by karawin