karawin/Ka-Radio32 #80: rotary encoder problem

gulalo, 2018-12-28:
Hello Jean-Pierre,
with 1.6R2 i have a problem selecting a station with encoder. (push&turn)
It does not work any longer, but IR is working. also very slow when adjusting output level with encoder. 
With 1.6R1 everything was  ok.

karawin, 2018-12-30:
Could you try with the R2 reloaded?

gulalo, 2018-12-31:
ok, reloaded R2
encoder works as follows:
short push and release: stops and starts playing
2 short push: time diaplay
push and hold, then turn: no change of Station possible (think it worked with R1)
push and hold untill Station is displayed, then release and turn: Station can be changed
is that what you  expect?
output Level adjusting works fine.
thank you, for this excellent projekt.
I wish you a happy and prosperous New Year 2019

karawin, 2018-12-31:
push and hold, then turn will work in the next release.
It is a regression due to code optimization.
Thank for your feedback.  and happy new year 2019 too.

karawin, 2018-12-31:
New R3 delivered
Happy new year 2019.

closed 2019-01-03 by karawin