karawin/Ka-Radio32 #81: Regression on WROVER (4Mbytes PSRAM version)

lyusupov, 2019-01-03:
Board: TTGO v1.8 (with ESP32-WROVER "classic")
**It works great with KaRadio32v1.5r5.bin** !

However every of upgrades onto 1.6 R0, R1, R2, R3... has failed.

This is an example of the failure (with "raw" stack) **with  1.6R0**:
```
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:3564
load:0x40078000,len:9372
ho 0 tail 12 room 4
load:0x40080400,len:5848
entry 0x400806d8
I (314) main: starting app_main()
I (314) main: RAM left: 4378640
E (324) main: Running partition type 0 subtype 17 (offset 0x001d0000)
I (364) main: Partition init done...
Log level is now ESP_LOG_ERROR
E (2194) vs1053: NO VS1053 detected
E (2194) main: LCD Type 5
X: 128, YY: 64, Y: 9
WIFI TRYING TO CONNECT TO SSID 1
E (4084) wifi: esp_wifi_disconnect 955 wifi not start
E (5684) event: invalid static ip
E (5684) main:
Wifi connected

DNS: 192.168.157.1
IP: 192.168.157.228

Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40130460  PS      : 0x00060730  A0      : 0x80131510  A1      : 0x3ffcf810
A2      : 0x3f8696e4  A3      : 0x3f869c84  A4      : 0x000001ff  A5      : 0x00000000
A6      : 0x000000bb  A7      : 0x00060023  A8      : 0x00060023  A9      : 0x3ffbc104
A10     : 0x00000000  A11     : 0x3ffc2770  A12     : 0x00000001  A13     : 0x00000001
A14     : 0x00060520  A15     : 0xffffffff  SAR     : 0x00000020  EXCCAUSE: 0x0000001d
EXCVADDR: 0x0000028b  LBEG    : 0x4009c15c  LEND    : 0x4009c18a  LCOUNT  : 0xffffffff

Backtrace: 0x40130460:0x3ffcf810 0x4013150d:0x3ffcf8e0 0x4013ac15:0x3ffcf900 0x4008d412:0x3ffcf950 0x400970b1:0x3ffcf990

Entering gdb stub now.
$T0b#e6
```
an this one with most recent binary of **1.6R3**:
```
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:3564
load:0x40078000,len:9372
ho 0 tail 12 room 4
load:0x40080400,len:5848
entry 0x400806d8
I (42) main: starting app_main()
I (42) main: RAM left: 4394264
E (52) main: Running partition type 0 subtype 17 (offset 0x001d0000)
I (82) main: Partition init done...
Log level is now ESP_LOG_ERROR
E (1862) vs1053: NO VS1053 detected
WIFI TRYING TO CONNECT TO SSID 1
E (3812) wifi: esp_wifi_disconnect 955 wifi not start
E (4712) event: invalid static ip

DNS: 192.168.157.1
IP: 192.168.157.228

CORRUPT HEAP: multi_heap.c:429 detected at 0x3f86aa08
abort() was called at PC 0x400978d7 on core 0

Backtrace: 0x4009227e:0x3ffafa80 0x400924f5:0x3ffafaa0 0x400978d7:0x3ffafac0 0x40084d6d:0x3ffafae0 0x40084e0f:0x3ffafb00 0x40089279:0x3ffafb50 0x40133cca:0x0

Entering gdb stub now.
$T0b#e6
```

Looked through history of closed tickets with no success.
Any clues on how to work this out ?

karawin, 2019-01-03:
Be sure to load the right csv bin file for the configuration.
Which one do you used?
lolin32pro.bin must be the right one

Send it here if it is yours.

lyusupov, 2019-01-03:
@karawin 
I use default one (standard_adb.bin) since it works fine to me with KaRadio32v1.5r5.bin:

```
$ esptool.py --chip esp32 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_freq 80m --flash_size detect 0x1000 bootloader.bin 0x8000 partitions.bin 0x10000 KaRadio32v1.5r5.bin 0x1D0000 KaRadio32v1.5r5.bin 0x3a2000 standard_adb.bin
```

This gives me nice output over UART:

```
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:3564
load:0x40078000,len:9372
ho 0 tail 12 room 4
load:0x40080400,len:5848
entry 0x400806d8
E (60) esp_core_dump: No core dump partition found!
I (61) main: starting app_main()
I (61) main: RAM left: 159680
E (61) main: Running partition type 0 subtype 17 (offset 0x001d0000)
I (101) main: Partition init done...
Log level is now ESP_LOG_ERROR
E (1911) vs1053: NO VS1053 detected
E (1911) main: LCD Type 5
X: 128, YY: 64, Y: 9
WIFI TRYING TO CONNECT TO SSID 1
E (3771) wifi: esp_wifi_disconnect 955 wifi not start
E (5111) event: invalid static ip
E (5111) main:
Wifi connected

DNS: 192.168.157.1
IP: 192.168.157.228

Init .......... Done
READY. Type help for a list of commands
```

And t**he radio works perfect**!

When I change main firmware binary argument(s) for esptool.py from KaRadio32v1.5r5.bin onto KaRadio32.bin :

```
esptool-2.5.1/esptool.py --chip esp32 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_freq 80m --flash_size detect 0x1000 bootloader.bin 0x8000 partitions.bin 0x10000 KaRadio32.bin 0x1D0000 KaRadio32.bin 0x3a2000 standard_adb.bin
```
- the system no longer boots up. Log was shown above.

Do you mean that TTGO T8's GPIO pins mapping conflicts with internal WROVER's PSRAM chip wiring ?

karawin, 2019-01-03:
Exactly

lyusupov, 2019-01-03:
@karawin 
Just in case I've tried lolinpro mapping - with no success. I2S sound was broken.
So I've unmapped GPIOs 16 and 17 from default layout:

```
$ diff standard_adb.csv ttgot8.csv
11,12c11,12
< P_ENC0_A,data,u8,16
< P_ENC0_B,data,u8,17
---
> P_ENC0_A,data,u8,255
> P_ENC0_B,data,u8,255
15c15
< P_ENC1_B,data,u8,17
---
> P_ENC1_B,data,u8,255
```
then I've build the binary and re-flashed it. It works good now!

I've rolled back comments to this ticket and see that you've updated your posts with new information.
I was not aware that you will decide to behave this way, so I've missed your updates initially.
Now I can see them. Thank you! I'll try out your binary as well.

lyusupov, 2019-01-03:
@karawin 

Could it make sense for you to :
- create a README.md in [binaries](https://github.com/karawin/Ka-Radio32/tree/master/binaries) folder of your repo and give advice there to use appropriate configuration binary for a particular ESP32 board ?
- add all available up to this moment configuration binaries there in addition to standard_adb.bin ?

karawin, 2019-01-03:
I made a boards directory for that with csv and bin in subdir built. Not described for the moment.
I will suggest to all the community to feed this directory for any kind of boards the found. I cannot do it myself.
Thanks for the suggestion of a md file. I will try, but the config is in fact in the csv.

closed 2019-01-03 by karawin