karawin/Ka-Radio32 #82: Topic for this repository

bazooka07, 2019-01-07:
May be you can add "I2S" as topic for your repository. 
And the next time "wrover" and  "bluetooth".

karawin, 2019-01-08:
If someone would help me to write it, I will be happy.
But i will try.

karawin, 2019-01-09:
Ok, i havn t catched that.

9 janv. 2019 a écrit :
>On the "code" page and under the title of the page, click on "manage
>topics" for opening an input box.
>At the end of this box, add one or more keywords separated by a white
>space. Finally click on the "Done"  button for saving. 

-- 
Envoyé de mon téléphone Android avec <a href="https://play.google.com/store/apps/details?id=com.onegravity.k10.pro2"><b>K-@ Mail</b></a>. Excusez la brièveté.

closed 2019-01-18 by karawin