karawin/Ka-Radio32 #84: Where  rda5807M support ?

alansbor, 2019-01-13:
Hello!

What are the reasons for removing the ability to use FM radio on rda5807M? I really want to use this opportunity.

karawin, 2019-01-18:
The reason is a lack of time. It was never released nor integrated and really tested.
[RDA5807.zip](https://github.com/karawin/Ka-Radio32/files/2772262/RDA5807.zip)

alansbor, 2019-01-22:
Ok. Files received. Thank a lot!
The question: How to use FM RDA if I have an SPI display and  SPI VS1053. 
Can I use the free pin of ESP32 drom I2S DAC or PDM output?

// I2S DAC or PDM output
//-----------------------
#define PIN_I2S_LRCK GPIO_NUM_25	// or Channel1
#define PIN_I2S_BCLK GPIO_NUM_26	// or channel2
#define PIN_I2S_DATA GPIO_NUM_22	//

karawin, 2019-01-22:
No. In the old days, I used the mic input of the vs1053 to mix the analog output of the rda to the output of the vs1053, but I removed that possibility.
On the other hand the command interface has never been done.
Sorry. No solution in KaRadio.

closed 2019-02-16 by karawin