karawin/Ka-Radio32 #87: Tricky conflict vs1053 and ili9341

pilnikov, 2019-01-26:
Hello JP.
My trickly pb:

my config:

key,type,encoding,value
gpio_space,namespace,,
K_SPI,data,u8,2
P_MISO,data,u8,19
P_MOSI,data,u8,23
P_CLK,data,u8,18
P_XCS,data,u8,15
P_RST,data,u8,12
P_XDCS,data,u8,2
P_DREQ,data,u8,4
P_ENC0_A,data,u8,255
P_ENC0_B,data,u8,255
P_ENC0_BTN,data,u8,255
P_ENC1_A,data,u8,255
P_ENC1_B,data,u8,255
P_ENC1_BTN,data,u8,255
P_BTN0_A,data,u8,255
P_BTN0_B,data,u8,255
P_BTN0_C,data,u8,255
P_BTN1_A,data,u8,255
P_BTN1_B,data,u8,255
P_BTN1_C,data,u8,255
P_I2C_SCL,data,u8,255
P_I2C_SDA,data,u8,255
P_I2C_RST,data,u8,255
P_LCD_CS,data,u8,13
P_LCD_A0,data,u8,14
P_LCD_RST,data,u8,12
P_IR_SIGNAL,data,u8,36
P_I2S_LRCK,data,u8,25
P_I2S_BCLK,data,u8,26
P_I2S_DATA,data,u8,22
P_TOUCH_CS,data,u8,255
P_JOY_0,data,u8,255
P_JOY_1,data,u8,255
P_LED_GPIO,data,u8,21
P_ADC_KBD,data,u8,255
P_BACKLIGHT,data,u8,27

option_space,namespace,,
O_LCD_TYPE,data,u8,194
O_LCD_ROTA,data,u8,255
O_LCD_OUT,data,u32,255
O_DDMM_FLAG,data,u8,1

i use ili9341 & vs1053

Doesn't work together.

if i send sys.lcd("0") work vs1053, but doesn't work (naturally 0<>194) ili9341
if i send sys.lcd("194") work ili9341, but doesn't work vs1053 (but work fine with i2s)

I assume that gpio is in conflict. Either somewhere, a part of spi(extended) is incorrectly initiated

karawin, 2019-01-31:
Normal, bad conf
P_LCD_RST,data,u8,12
_LCD_RST,data,u8,12
in conflict
Just assign another gpio for one of thse two.

pilnikov, 2019-01-31:
Solved

closed 2019-01-31 by pilnikov