karawin/Ka-Radio32 #89: Newbie help...

Asxetos21, 2019-02-02:
I am a total newbie on esp32, and i am trying to set this up, trying with 2 different displays.
1st option, an oled 128x64 ssd1306 on i2c.
2nd option, a 160x128 ST7735 on spi.
I couldn't get anyone to work (both work on other projects).
Reading the docs i changed what i figured needed changing but no luck.

Can someone tell me exactly which files need to be changed for each option to work?

Another question, in the images folder the "downloadtool32.jpg" shows the main bin (karadio32.bin) should be flashed in 2 different locations. Is it correct or is it a mistake?

Thanks...

Asxetos21, 2019-02-02:
Just an update,
All my attempts until now were new build from source. Compiling is successful, but the esp32 does not start.
I thought i should test with the ready made binaries, and the board starts ok.
The i2c display works sort of (shifted a couple of pixels).
The spi display does not.

I guess it comes to the same point. How to change the default displays, and why build from source does not work.

pepelnyy, 2019-02-04:
telnet to your Ka-Radio32 ip and enter: sys.lcd("x") where x is number corresponding your lcd type:
LCD_I2C_SSD1306		1 //128X64
LCD_I2C_SSD1306NN	4 //128X64
LCD_I2C_SSD1306UN	6 //128x32

LCD_SPI_ST7735			192 // 128x160  (LCD_COLOR|LCD_SPI =0xC0)
LCD_SPI_ST7735S			200 // 128x128
LCD_SPI_ST7735L			201 // 80x160
LCD_SPI_ST7735W			202 // 128x160 shifted 2+1

Asxetos21, 2019-02-05:
My setup with Ka-Radio32 works.
Playing stations works fine (vs1053), I can access the web interface, telnet into it etc... Only the screen is not initialized.
I triple-checked everything, tested all 4 ST7735 lcd codes, checked that indeed the lcd type was changed, but still...

The spi lcd i have is probably one of many Chinese knockoffs, but even so it works correctly with Edzefs esp32-radio Arduino project, with Adafruit_ST7735 library if i declare it as ST7735R "blacktab" type.

It also works perfectly on Arduino-IDE with the TFT_eSPI library examples, if i declare it as ILI9163.

Any advise on how i can troubleshoot it?

pilnikov, 2019-02-05:
Try an example from this https://github.com/olikraus/Ucglib_Arduino library in the Arduino IDE. Modify the example for use with your scheme (do not change the scheme, change the program!).

Asxetos21, 2019-02-05:
My bad, going back and forth between Edzelf's ESP-Radio and Karawin's Ka-Radio32 trying to troubleshoot, i reversed 2 lcd pins. I saw it on the xxxth check. The lcd works fine as an ST7735.
Thanks for the help and sorry, closing issue.

Wonderful project, most stable so far. A big plus for me is support for Greek characters.

closed 2019-02-05 by Asxetos21