karawin/Ka-Radio32 #93: Problems with 1.8R1

Asxetos21, 2019-02-07:
Update to 1.8R1 from the web interface causes loss of IR control, LCD corruption and finally freezing.
Esp32 dev board (15+15 pin version) VS1053, ST7735, rotary enc. and IR.
Haven't tested it any further...

Going back to 1.8R0, all OK again.

karawin, 2019-02-09:
No reason.
Try again

Elbskipper, 2019-02-10:
Same issue with 1.8R1. Display is often corrupted after Power Up (just snow). Chip: ST7735,  128x160 .
With older versions it was always OK. Config: Esp32 dev board, VS1053, Rotary, ST7735.

karawin, 2019-02-10:
Check with sys.lcd command the type of lcd.

Elbskipper, 2019-02-11:
This was the first thing I did. I even changed to another display type and back without success. The strange thing is, that it sometimes works. The web interface is always ok, even if the display has failed.

karawin, 2019-02-11:
May be a conflict in the definition of gpio in the csv file (check carefully if it is your definitions or which one do you use?) or a problem of sequencing in the init.
I don't have the problem so it is hard to find.

Elbskipper, 2019-02-11:
I am using your preconfigured binary’s.
 
I will recheck the wiring and maybe I will reprogram the entire system. I somehow got the feeling (sorry for this word) that an issue was introduced, when you modified the display binary’s, I think it was in one of the updates of the 1.7 release.

karawin, 2019-02-11:
Which csv if any please? Or the bare KaRadio32 bin only

Elbskipper, 2019-02-11:
Just the bare KaRadio32.bin. No individual compilations. After the flashing I set the display to type “192”. 
 
In the meantime I re-flashed the entire system via the ESP32 Download Tool. I couldn`t reproduce the issue since. It looks like something went wrong during one of the OTA updates.

Elbskipper, 2019-02-12:
I`ve repowered the System at least 20 times since the re-flash. No more display issues. So I consider this issue as fixed.

karawin, 2019-02-12:
ok, closed

closed 2019-02-12 by karawin