karawin/Ka-Radio32 #94: KaRadio32 does not connect to Wifi after upgrade to 1.8x

phanus, 2019-02-15:
My KaRadio32 worked fine with release 1.6R4. Then I performed OTA to 1.8 and it is not able connect to my WiFi. I can connect to KaRadio's AP at 192.168.4.1 and when I input my WiFi SSID and PASS KaRadio32 restarts again into AP mode, unable to connect to my WiFi.
If I downgrade back to 1.6R4 via ESP32 Flasher it works perfectly again.

I'm using ESP32 WROVER with 4MB PSRAM and VS1053.

Below is verbose log of the boot process, line "E (5707) wifi: esp_wifi_disconnect 964 wifi not start" looks suspicious.

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3564
load:0x40078000,len:9372
ho 0 tail 12 room 4
load:0x40080400,len:5848
entry 0x400806d8
I (107) main: starting app_main()
I (107) main: RAM left: 4394720
E (117) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (137) main: Partition init done...
D (227) eeprom: saveDeviceSettings
Log level is now ESP_LOG_VERBOSE
V (337) addon: LINE: Log level is now ESP_LOG_VERBOSE

I (447) vs1053: VS1053 LFreq: 1600000
I (447) vs1053: VS1053 HFreq: 6153846
I (3747) vs1053: VS1053/VS1003 detected. MP3Status: 40, Version: 4
I (3747) vs1053: SCI_Status  = 0x40
I (3747) vs1053: SCI_Mode (0x4800) = 0x4802
I (3747) vs1053: SCI_ClockF = 0xB800
I (3747) vs1053: VS Version (VS1053 is 4) = 4
I (3747) vs1053: I2S Speed: 0
I (3757) VS1053 Patches: plugin size 8896 4448
I (3757) VS1053 Patches: plugin start: 7 1 8050 6

I (4087) vs1053: volume: 254
I (4087) main: hardware initialized
I (4087) main: Hardware init done...
I (4107) main: LCD Type 0
D (4167) addonu8g2: lcd init BW type: 0
D (4167) u8g2_hal: gpio_and_delay_cb: Received a msg: 40, arg_int: 0, arg_ptr: 0x0
I (4167) u8g2_hal: sda_io_num 13
I (4167) u8g2_hal: scl_io_num 14
I (4177) u8g2_hal: clk_speed 90000
I (4177) u8g2_hal: i2c_param_config 1
I (4177) u8g2_hal: i2c_driver_install 0
D (4187) u8g2_hal: gpio_and_delay_cb: Received a msg: 75, arg_int: 1, arg_ptr: 0x0
D (4187) u8g2_hal: gpio_and_delay_cb: Received a msg: 41, arg_int: 100, arg_ptr: 0x0
D (4297) u8g2_hal: gpio_and_delay_cb: Received a msg: 75, arg_int: 0, arg_ptr: 0x0
D (4297) u8g2_hal: gpio_and_delay_cb: Received a msg: 41, arg_int: 100, arg_ptr: 0x0
D (4397) u8g2_hal: gpio_and_delay_cb: Received a msg: 75, arg_int: 1, arg_ptr: 0x0
D (4397) u8g2_hal: gpio_and_delay_cb: Received a msg: 41, arg_int: 100, arg_ptr: 0x0
I (4547) addonu8g2: X: 128, YY: 64, Y: 9

I (5557) main: Set Song buffer to 420k
I (5557) main: audio_output_mode 4
One of I2S=0, I2S_MERUS, DAC_BUILT_IN, PDM, VS1053
I (5557) main: Set baudrate at 115200
I (5557) main: Release 1.8, Revision 1
I (5557) main: SDK v3.3-beta1-268-g5c88c5996-dirty
I (5567) main: Heap size: 3954460
I (5617) main: Volume set to 254
I (5647) main: t0 task: 3ffb1b18
I (5647) main: starting wifi
I (5647) wifi: wifi driver task: 3ffbd900, prio:23, stack:3584, core=0
I (5647) wifi: wifi firmware version: 38e2484
I (5647) wifi: config NVS flash: enabled
I (5657) wifi: config nano formating: disabled
I (5677) wifi: Init dynamic tx buffer num: 32
I (5677) wifi: Init data frame dynamic rx buffer num: 32
I (5677) wifi: Init management frame dynamic rx buffer num: 32
I (5677) wifi: Init static tx buffer num: 16
I (5687) wifi: Init static rx buffer size: 1600
I (5687) wifi: Init static rx buffer num: 10
I (5687) wifi: Init dynamic rx buffer num: 32
D (5697) eeprom: saveDeviceSettings
WIFI TRYING TO CONNECT TO SSID 1
E (5707) wifi: esp_wifi_disconnect 964 wifi not start
I (5737) main: connecting
I (5787) wifi: mode : sta (30:ae:a4:08:98:58)
I (5797) main: Initialised wifi
I (8207) wifi: new:<2,1>, old:<1,0>, ap:<255,255>, sta:<2,1>, prof:1
I (8207) wifi: state: init -> auth (b0)
I (8207) wifi: state: auth -> assoc (0)
I (8217) wifi: state: assoc -> run (10)
I (8227) wifi: state: run -> init (ea0)
I (8227) wifi: new:<2,0>, old:<2,1>, ap:<255,255>, sta:<2,1>, prof:1
E (8227) main: 
Wifi Disconnected. Connection tried again
E (8237) main: 
Wifi Disconnected.
E (10647) main: 
Wifi Disconnected. Connection tried again
E (10647) main: 
Wifi Disconnected.
I (13067) wifi: new:<2,1>, old:<2,0>, ap:<255,255>, sta:<2,1>, prof:1
I (13067) wifi: state: init -> auth (b0)
I (13067) wifi: state: auth -> assoc (0)
I (13077) wifi: state: assoc -> run (10)
I (13087) wifi: state: run -> init (ea0)
I (13087) wifi: new:<2,0>, old:<2,1>, ap:<255,255>, sta:<2,1>, prof:1
E (13087) main: 
Wifi Disconnected. Connection tried again
E (13087) main: 
Wifi Disconnected.
E (15507) main: 
Wifi Disconnected. Connection tried again
E (15507) main: 
Wifi Disconnected.

karawin, 2019-02-15:
You are on a wrover cpu with 4mB of psram.
Partitions have changed.
Please reload all bin and flash directly.

phanus, 2019-02-16:
I flashed latest files from the Karadio32/Binaries folder (except the HW config file which I created new from scratch just to be sure) and ended up with the same problem. It does not connect to WiFi.
(And yes in between the restarts in the log, I connected to AP and I set up the SSID/Pass)
New log is below, together with my HW csv.

LOG--------------------------------------------
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3564
load:0x40078000,len:9372
ho 0 tail 12 room 4
load:0x40080400,len:5848
entry 0x400806d8
I (110) main: starting app_main()
I (110) main: RAM left: 4395112
E (120) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (160) main: Partition init done...
E (160) main: Device config not ok. Try to restore
E (210) main: Device config not cleared. Clear it.
#erase All done##
#erase All done##
#erase All done##
#erase All done##
#erase All done##
#erase All done##
V (560) addon: LINE: #erase All done##

D (560) eeprom: saveDeviceSettings
Log level is now ESP_LOG_ERROR
WIFI TRYING TO CONNECT TO SSID 255
E (5960) main: The default AP is  WifiKaRadio. Connect your wifi to it.
Then connect a webbrowser to 192.168.4.1 and go to Setting
May be long to load the first time.Be patient.
E (6310) main: mDNS Hostname: karadio32
READY. Type help for a list of commands
E (30680) event: mismatch or invalid event, id=63
E (30680) event: default event handler failed!
dhcps: send_offer>>udp_sendto result 0
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3564
load:0x40078000,len:9372
ho 0 tail 12 room 4
load:0x40080400,len:5848
entry 0x400806d8
I (110) main: starting app_main()
I (110) main: RAM left: 4395112
E (120) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (160) main: Partition init done...
Log level is now ESP_LOG_ERROR
WIFI TRYING TO CONNECT TO SSID 1
E (5450) wifi: esp_wifi_disconnect 964 wifi not start
E (7940) main: 
Wifi Disconnected. Connection tried again
E (7950) main: 
Wifi Disconnected.
E (10360) main: 
Wifi Disconnected. Connection tried again
E (10360) main: 
Wifi Disconnected.
E (12790) main: 
Wifi Disconnected. Connection tried again
E (12790) main: 
Wifi Disconnected.
E (15200) main: 
Wifi Disconnected. Connection tried again
E (15200) main: 
Wifi Disconnected.
E (17640) main: 
Wifi Disconnected. Connection tried again
E (17640) main: 
Wifi Disconnected.
E (20060) main: 
Wifi Disconnected. Connection tried again
E (20060) main: 
Wifi Disconnected.
E (22490) main: 
Wifi Disconnected. Connection tried again
E (22490) main: 
Wifi Disconnected.
E (24900) main: 
Wifi Disconnected. Connection tried again
E (24900) main: 
Wifi Disconnected.

device->current_ap: 2
WIFI TRYING TO CONNECT TO SSID 2
E (25520) wifi: esp_wifi_disconnect 964 wifi not start

device->current_ap: 0
WIFI GO TO AP MODE
E (45560) main: The default AP is  WifiKaRadio. Connect your wifi to it.
Then connect a webbrowser to 192.168.4.1 and go to Setting
May be long to load the first time.Be patient.
E (45770) main: mDNS Hostname: karadio32
READY. Type help for a list of commands

CSV-----------------------------------------
key,type,encoding,value
gpio_space,namespace,,
K_SPI,data,u8,2
P_MISO,data,u8,19
P_MOSI,data,u8,23
P_CLK,data,u8,18
P_XCS,data,u8,32
P_RST,data,u8,12
P_XDCS,data,u8,33
P_DREQ,data,u8,34
P_ENC0_A,data,u8,255
P_ENC0_B,data,u8,255
P_ENC0_BTN,data,u8,255
P_ENC1_A,data,u8,15
P_ENC1_B,data,u8,27
P_ENC1_BTN,data,u8,5
P_BTN0_A,data,u8,255
P_BTN0_B,data,u8,255
P_BTN0_C,data,u8,255
P_BTN1_A,data,u8,255
P_BTN1_B,data,u8,255
P_BTN1_C,data,u8,255
P_I2C_SCL,data,u8,14
P_I2C_SDA,data,u8,13
P_I2C_RST,data,u8,2
P_LCD_CS,data,u8,13
P_LCD_A0,data,u8,14
P_LCD_RST,data,u8,2
P_IR_SIGNAL,data,u8,255
P_I2S_LRCK,data,u8,25
P_I2S_BCLK,data,u8,26
P_I2S_DATA,data,u8,22
P_TOUCH_CS,data,u8,255
P_JOY_0,data,u8,255
P_JOY_1,data,u8,255
P_LED_GPIO,data,u8,4
P_ADC_KBD,data,u8,255
P_BACKLIGHT,data,u8,255

option_space,namespace,,
O_LCD_TYPE,data,u8,192
O_LCD_ROTA,data,u8,0
O_LCD_OUT,data,u32,0
O_DDMM_FLAG,data,u8,0

custom_ir_space,namespace,,
K_UP,data,string,
K_LEFT,data,string,
K_OK,data,string,0
K_RIGHT,data,string,
K_DOWN,data,string,
K_0,data,string,
K_1,data,string,
K_2,data,string,
K_3,data,string,
K_4,data,string,
K_5,data,string,
K_6,data,string,
K_7,data,string,
K_8,data,string,
K_9,data,string,
K_STAR,data,string,
K_DIESE,data,string,
K_INFO,data,string,

karawin, 2019-02-16:
Ok, A strange pb on wifi.
A good start is
Log level is now ESP_LOG_INFO
I (419) vs1053: VS1053 LFreq: 1403508
I (419) vs1053: VS1053 HFreq: 6153846
I (6709) wifi: state: init -> auth (b0)
I (6709) wifi: state: auth -> assoc (0)
I (6719) wifi: state: assoc -> run (10)
I (6729) wifi: connected with karawin2, channel 9, bssid = 00:24:d4:d5:81:60
I (6729) wifi: pm start, type: 1

In AP mode, can you type the command 
sys.logd
in serial terminal and reboot?

karawin, 2019-02-16:
Some limits on length ssid name and password
#define SSIDLEN		32
#define PASSLEN		60 
#define HOSTLEN		24

phanus, 2019-02-16:
My pass was over 60 chars… I reconfigured my router and now the KaRadio32 connected successfully. Might be worth supporting full 63 chars according to WiFi standard for AES PSK.
Thank you for your excellent support!

karawin, 2019-02-16:
Yes, it was 64, but for bad reason I changed it to 60. I will see to go back to 64.
Sorry.

phanus, 2019-02-16:
OTA to 1.9 R0 connected OK with former (longer) pass. :)

karawin, 2019-02-16:
Thanks for the donation. I close this issue.

closed 2019-02-16 by karawin