karawin/Ka-Radio32 #95: Rotary encoder

nano111, 2019-02-15:
Hi,
I  bought few encoders from China.They are 20 notches all around and they do not work
properly.With one or two turns in one direction it advances station,but if I go further in the same direction it reverses back one station and stops.Same in the other directions.Somitimes just activates station changing but does not change them.I have checked with 3 of them and everything is the same.I have one with 30 notches (positions 360) and that work OK,
Is there an easy fix for this?Tried sys.henc1 with 1 and 0 -no luck.
Thanks

karawin, 2019-02-15:
I had the same problem.
Wait for the next release or try the prerelease
sys.prerelease on serial or telnet interface. 
May be an inversion of A B is needed.

nano111, 2019-02-16:
Thanks , I will try it.
Changing A and B on the one that works only changed direction
of setting,but enything is possible :-)
Regards

nano111, 2019-02-16:
It is working much better with 1.9 R0,still skips few turns ,and sometmes goes back one
 station.
Maybe some more fine tuning.
Thanks

karawin, 2019-02-16:
I will try with a sampling at 500µs in place of 1ms
You will test it soon in a prerelease.

karawin, 2019-02-16:
You can test the prerelease.
sys.prerelease
Thanks

nano111, 2019-02-17:
This is where I bought them
https://m.aliexpress.com/item/32864246557.html?pid=808_0000_0101&spm=a2g0n.search-amp.list.32864246557&aff_trace_key=e0d67521a34d45c1b5272eed99fd3b93-1549920695374-04952-b67aVwFq&aff_platform=msite&m_page_id=33amp-3txjCJNNdNrUJdt9sbvCZQ1550390259305
If you give me your address , I will send you one as a gift.
Thanks, will try prerelease

karawin, 2019-02-17:
Ok, it is a bare rotary.
Do you used the pull up?
[image]
Thanks for the proposal but don't worry about that.

nano111, 2019-02-17:
Tried pererelease,it is worse with henc 0 but the same with henc1.Sometimes it does not react on turning .Seems like it goes back much better than forward.
Tried pull ups before with 1.8 and no difference.Iguess these gpios have software pullup.

nano111, 2019-02-17:
Connected pull up resistors and now it works perfectly.
Thanks

nano111, 2019-02-17:
Did you consider automaticaly switching to clock after few minutes of inactivity?
In that case you can leave this radio to stay on all the time and use it as a clock ,just switch amplifier off
or stero system if it is connected to one.

karawin, 2019-02-17:
I am thinking of staying on the clock screen by default in case of double click.
For now, it display the clock but any event revert back to the main screen.
Good news for the encoders. It seems that the internal pull up are not enough. Good to know.
Thanks.

closed 2019-02-18 by nano111