karawin/Ka-Radio32 #96: Release 1.9R0, Problem with VS1053

Elbskipper, 2019-02-17:
HI,
since I did an OTA update to 1.9R0 there is no sound anymore via the VS1053 board.
If I unload the patch (sys.patch("0") it is ok. If I load it again it fails again.
In the past I always used the patch.
My config:
VS1053, Rotary Encoder, LCD_SPI_ST7735
No customized config file

karawin, 2019-02-19:
Please OTA again, and send the log in level i (sys.logi)

Elbskipper, 2019-02-19:
OK, did OTA again. Same Problem with patch loaded.
 
Attached are two logfiles. One with and one without patch.

 

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3580
load:0x40078000,len:9300
load:0x40080400,len:5764
entry 0x400806d4
E (572) spiram: SPI RAM enabled but initialization failed. Bailing out.
I (136) main: starting app_main()
I (136) main: RAM left: 200488
E (136) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (156) main: Partition init done...
Log level is now ESP_LOG_INFO
I (366) vs1053: VS1053 LFreq: 1403508
I (366) vs1053: VS1053 HFreq: 6153846
I (4266) vs1053: VS1053/VS1003 detected. MP3Status: 40, Version: 4
I (4266) vs1053: SCI_Status  = 0x40
I (4266) vs1053: SCI_Mode (0x4800) = 0x4802
I (4266) vs1053: SCI_ClockF = 0xB800
I (4266) vs1053: VS Version (VS1053 is 4) = 4
I (4266) vs1053: I2S Speed: 0
I (4276) VS1053 Patches: plugin size 8896 4448
I (4276) VS1053 Patches: plugin start: 7 1 8050 6

I (4616) vs1053: volume: 42
I (4616) main: hardware initialized
I (4616) main: Hardware init done...
I (4636) main: LCD Type 192
I (4666) addonucg: lcd init  type: 192, Rotat: 0
I (4716) ucg_hal: UCG_COM_MSG_POWER_UP: 80 ns so in Hz: 12500000
I (4716) ucg_hal: ... Adding spi lcd bus  Speed= 25000000.
I (4836) addonucg: lcd init Color type: 192
I (4896) addonucg: X: 160, yy: 128, y: 10

I (4906) main: audio_output_mode 4
One of I2S=0, I2S_MERUS, DAC_BUILT_IN, PDM, VS1053
I (4906) main: Set Song buffer to 50k
I (4906) main: Set baudrate at 115200
I (4906) main: Release 1.9, Revision 0
I (4906) main: SDK v3.3-beta1-268-g5c88c5996-dirty
I (4916) main: Heap size: 139020
I (5266) main: Volume set to 42
I (5296) main: t0 task: 3ffb1a88
I (5296) main: starting wifi
I (5306) wifi: wifi driver task: 3ffbd558, prio:23, stack:3584, core=0
I (5306) wifi: wifi firmware version: 38e2484
I (5306) wifi: config NVS flash: enabled
I (5306) wifi: config nano formating: disabled
I (5326) wifi: Init dynamic tx buffer num: 32
I (5326) wifi: Init data frame dynamic rx buffer num: 32
I (5326) wifi: Init management frame dynamic rx buffer num: 32
I (5326) wifi: Init static tx buffer num: 16
I (5336) wifi: Init static rx buffer size: 1600
I (5336) wifi: Init static rx buffer num: 10
I (5346) wifi: Init dynamic rx buffer num: 32
WIFI TRYING TO CONNECT TO SSID 2
E (5346) wifi: esp_wifi_disconnect 964 wifi not start
I (5366) main: connecting
I (5426) wifi: mode : sta (3c:71:bf:8c:01:78)
I (5426) main: Initialised wifi
[... 934 lines cut ...]
READY. Type help for a list of commands
##CLI.ICY0#: at IP 192.168.178.58
autostart: playing:1, currentstation:5
##CLI.NAMESET#: 5 Radio Paloma
##CLI.URLSET#: www.radiopaloma.de
##CLI.PATHSET#: /listen.pls
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 42
I (9687) webserver: playstationInt: 5, g_device: 5
I (10237) webclient: Webclient socket: 55, errno: 0
##CLI.URLSET#: pool.radiopaloma.de
##CLI.PATHSET#: /RADIOPALOMA.mp3
##CLI.PORTSET#: 80
##CLI.STOPPED# from C_PLIST
I (11537) webclient: watermark : 3b4  948
I (11537) webclient: Webclient socket: 57, errno: 0
##CLI.ICY0#: Radio Paloma - 100% Deutscher Schlager!
##CLI.ICY3#: http:\/\/www.radiopaloma.de
##CLI.ICY4#: Schlager
##CLI.ICY5#: 128
##CLI.ICY6#: 100% Deutscher Schlager!
##CLI.PLAYING#
##CLI.META#: Nik P. - Der Fremde
##CLI.META#: Ela - Universum

karawin, 2019-02-19:
Every things are normal seen from the log.

- You need to cut the local echo to avoid #INFO:" lines. and #INFO:"#INFO:" etc. (never ends)

- The station or the wifi stream is very low.
I (1794436) audio_player: Buffer fill 2%, 1436 // 51200 bytes 
I (1813266) audio_player: Buffer fill 3%, 1848 // 51200 bytes
I (1832026) audio_player: Buffer fill 1%, 971 // 51200 bytes 
I (1851166) audio_player: Buffer fill 2%, 1436 // 51200 bytes
....
Is abnormal.

Elbskipper, 2019-02-20:
OK,
I double checked the buffer fill rate. With the patch loaded it is always between 2 and 4%, without the patch it is 90 to 94%. I checked it with various stations.
I remember, that there was a necessary hardware modification required to the decoder board (short pins 33 and 34). Is this still necessary?
 
Is there any way to reload the previous release (i.e. 1.7..)?

karawin, 2019-02-20:
I remember, that there was a necessary hardware modification required to the decoder board (short pins 33 and 34). Is this still necessary?
No it was a fake news because in KaRadio a patch is included to avoid it.

The vs1053 patch cannot explain your problem. It seems related to the echo in the serial. The cpu is fulled by the serial read. A kind of larsen.

Elbskipper, 2019-02-20:
Here is a link with the description of the issue people had some time ago:
 
 <http://www.bajdi.com/lcsoft-vs1053-mp3-module/> http://www.bajdi.com/lcsoft-vs1053-mp3-module/
 
But I do not think, that this is my issue.
 
Here is a new logfile with patch. No more echo’s, but still no sound.
 
I (194747) wifi: state: run -> init (0)
I (194747) wifi: pm stop, total sleep time: 89818224 us / 188996938 us
 
I (194747) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (194767) wifi: flush txq
I (194767) wifi: stop sw txq
I (194767) wifi: lmac stop hw txq
E (194767) main: Wifi Disconnected.
ets Jun  8 2016 00:22:57
 
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3580
load:0x40078000,len:9300
load:0x40080400,len:5764
entry 0x400806d4
E (573) spiram: SPI RAM enabled but initialization failed. Bailing out.
I (137) main: starting app_main()
I (137) main: RAM left: 200488
E (137) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (157) main: Partition init done...
Log level is now ESP_LOG_INFO
I (367) vs1053: VS1053 LFreq: 1403508
I (367) vs1053: VS1053 HFreq: 6153846
I (4267) vs1053: VS1053/VS1003 detected. MP3Status: 40, Version: 4
I (4267) vs1053: SCI_Status  = 0x40
I (4267) vs1053: SCI_Mode (0x4800) = 0x4802
I (4267) vs1053: SCI_ClockF = 0xB800
I (4267) vs1053: VS Version (VS1053 is 4) = 4
I (4267) vs1053: I2S Speed: 0
I (4277) VS1053 Patches: plugin size 8896 4448
I (4277) VS1053 Patches: plugin start: 7 1 8050 6
 
I (4617) vs1053: volume: 35
I (4617) main: hardware initialized
I (4617) main: Hardware init done...
I (4637) main: LCD Type 192
I (4667) addonucg: lcd init  type: 192, Rotat: 0
I (4717) ucg_hal: UCG_COM_MSG_POWER_UP: 80 ns so in Hz: 12500000
I (4717) ucg_hal: ... Adding spi lcd bus  Speed= 25000000.
I (4837) addonucg: lcd init Color type: 192
I (4897) addonucg: X: 160, yy: 128, y: 10
 
I (4907) main: audio_output_mode 4
One of I2S=0, I2S_MERUS, DAC_BUILT_IN, PDM, VS1053
I (4907) main: Set Song buffer to 50k
I (4907) main: Set baudrate at 115200
I (4907) main: Release 1.9, Revision 0
I (4907) main: SDK v3.3-beta1-268-g5c88c5996-dirty
I (4917) main: Heap size: 139020
I (5267) main: Volume set to 35
I (5297) main: t0 task: 3ffb1a88
I (5297) main: starting wifi
I (5307) wifi: wifi driver task: 3ffbd558, prio:23, stack:3584, core=0
I (5307) wifi: wifi firmware version: 38e2484
I (5307) wifi: config NVS flash: enabled
I (5307) wifi: config nano formating: disabled
I (5327) wifi: Init dynamic tx buffer num: 32
I (5327) wifi: Init data frame dynamic rx buffer num: 32
I (5327) wifi: Init management frame dynamic rx buffer num: 32
I (5327) wifi: Init static tx buffer num: 16
I (5337) wifi: Init static rx buffer size: 1600
I (5337) wifi: Init static rx buffer num: 10
I (5337) wifi: Init dynamic rx buffer num: 32
WIFI TRYING TO CONNECT TO SSID 2
E (5347) wifi: esp_wifi_disconnect 964 wifi not start
I (5367) main: connecting
I (5427) wifi: mode : sta (3c:71:bf:8c:01:78)
I (5427) main: Initialised wifi
I (5557) wifi: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (6227) wifi: state: init -> auth (b0)
I (6237) wifi: state: auth -> assoc (0)
I (6237) wifi: state: assoc -> run (10)
I (6257) wifi: connected with HomeLan, channel 11, bssid = c8:0e:14:f3:c4:47
I (6257) wifi: pm start, type: 1
 
E (6257) event: invalid static ip
I (6257) main: Wifi connected
 
DNS: 192.168.178.1
IP: 192.168.178.58
 
I (7147) main: mDNS Init ok
E (7147) main: mDNS Hostname: karadio32
I (7147) renderer: chip revision 1
I (7177) renderer: i2s intr:0
I (7517) main: RAM left 69184
I (7517) main: uartInterfaceTask task: 3ffb8b50
I (7527) main: clientTask task: 3ffb8cc8
I (7537) servers: watermark: 0xab8  2744
I (7537) servers: telnetServer_sock socket: 52, errno: 0
I (7537) servers: Webserver socket: 53, errno: 0
I (7537) main: serversTask task: 3ffe91a8
I (7547) main: task_addon task: 3ffea0c4
W (7667) GPIO: g_get_adc err 0x1102
I (7667) addon: rmt_nec_rx_task task: 3ffeac44
I (7667) addon: task_lcd task: 3ffeb648
I (8147) main:  Init Done
READY. Type help for a list of commands
##CLI.ICY0#: at IP 192.168.178.58
autostart: playing:1, currentstation:10
##CLI.NAMESET#: 10 Hotmix Rock
##CLI.URLSET#: streamingads.hotmixradio.fm
##CLI.PATHSET#: /hotmixradio-rock-128.mp3
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 35
I (9997) webserver: playstationInt: 10, g_device: 10
I (10527) webclient: Webclient socket: 56, errno: 0
##CLI.ICY0#: Hot Mix Radio ROCK
##CLI.ICY3#: http:\/\/www.hotmixradio.fr\/rock
##CLI.ICY4#: rock
##CLI.ICY5#: 128
##CLI.PLAYING#
##CLI.META#: BAND OF HORSES - The Funeral
I (19487) audio_player: Buffer fill 2%, 1436 // 51200 bytes
I (36657) audio_player: Buffer fill 3%, 1848 // 51200 bytes
I (55557) audio_player: Buffer fill 2%, 1149 // 51200 bytes
I (74657) audio_player: Buffer fill 3%, 1848 // 51200 bytes
##CLI.META#: OLIVER TREE - Hurt
I (93357) audio_player: Buffer fill 3%, 1847 // 51200 bytes
I (112287) audio_player: Buffer fill 1%, 527 // 51200 bytes
I (131347) audio_player: Buffer fill 2%, 1435 // 51200 bytes
I (150117) audio_player: Buffer fill 3%, 1848 // 51200 bytes
I (168987) audio_player: Buffer fill 3%, 1648 // 51200 bytes
I (188077) audio_player: Buffer fill 3%, 1848 // 51200 bytes
I (206907) audio_player: Buffer fill 1%, 526 // 51200 bytes
I (226067) audio_player: Buffer fill 3%, 1848 // 51200 bytes
##CLI.META#: PALE WAVES - One More Time
I (245657) audio_player: Buffer fill 3%, 1847 // 51200 bytes
I (264397) audio_player: Buffer fill 2%, 1029 // 51200 bytes
I (283237) audio_player: Buffer fill 3%, 1848 // 51200 bytes
I (301977) audio_player: Buffer fill 3%, 1848 // 51200 bytes
I (320747) audio_player: Buffer fill 2%, 1436 // 51200 bytes
I (339937) audio_player: Buffer fill 3%, 1848 // 51200 bytes

karawin, 2019-02-20:
Can't see any reason for not receiving the stream with or without patch.
The only reason i can imagine is a bad DREQ request (permanent level 1) that read permanently the audio buffer.
The patch is from vlsi, so i cannot check it.
http://www.vlsi.fi/fileadmin/software/VS10XX/vs1053b-patches.pdf
But hundred of users use it.

Please check the DREQ wire.

Elbskipper, 2019-02-20:
I checked the wiring. It looks OK.
I just wonder, why it worked until the last OTA.
Anyway, I can live without the patch.

Thanks for your support.

karawin, 2019-02-20:
I think that it is a vs1053 crash. From 1.5 or 1.6, the patch is the new one. May be a consequence?
But never heard about problem before. May be a power supply a bit weak?
Bye

Elbskipper, 2019-02-22:
Sorry,
it's me again…..
I really interested in why I do have this issue. So I installed the vs1053 to my MCU8255 board. I loaded Software Version V1.7R1. Next I updated to V1.8R2. With both versions it worked fine with the patch loaded.
Next I connected the vs1053 back to the ESO32 board. Then I loaded Software Version V1.5R2. With this Version it workes fine with the patch loaded.

Here is the log file:

`
#WIFI.PATCH#ch is not loaded#
VS1053 Patch will be Loaded after power Off and On#
##WIFI.PATCH#
I (50351) wifi: state: run -> init (0)
I (50361) wifi: pm stop, total sleep time: 23120797 us / 44051042 us

I (50361) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (50381) wifi: flush txq
I (50381) wifi: stop sw txq
I (50381) wifi: lmac stop hw txq
E (50381) main:
Wifi Disconnected. Connection tried again
E (50381) main:
Wifi Disconnected. reboot
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3580
load:0x40078000,len:9300
load:0x40080400,len:5764
entry 0x400806d4
E (130) esp_core_dump: No core dump partition found!
I (132) main: starting app_main()
I (132) main: RAM left: 159656
E (132) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (172) main: Partition init done...
Log level is now ESP_LOG_INFO
E (4212) vs1053: VS1053/VS1003 detected. MP3Status: 40, Version: 4
I (4212) vs1053: SCI_Status  = 0x40
I (4212) vs1053: SCI_Mode (0x4800) = 0x4802
I (4212) vs1053: SCI_ClockF = 0xB800
I (4212) vs1053: VS Version (VS1053 is 4) = 4
device: 3ffcd380
I2S Speed: 0
I (4222) VS1053 Patches: plugin size 6350 3175
I (4222) VS1053 Patches: plugin start: 7 1 8050 6

I (4442) VS1053 Patches: plugin size 234 117
I (4442) VS1053 Patches: plugin start: 7 1 8f00 6

volume: 35
I (4542) main: hardware initialized
I (4542) main: Hardware init done...
E (4542) main: LCD Type 192
I (4542) addonucg: lcd init  type: 192
I (4632) ucg_hal: UCG_COM_MSG_POWER_UP: 100 ns so in Hz: 10000000
I (4632) ucg_hal: ... Adding device bus  Speed= 20000000.
I (4702) addonucg: lcd init Color type: 192
X: 160, yy: 128, y: 14
I (4832) main: audio_output_mode 4
One of I2S=0, I2S_MERUS, DAC_BUILT_IN, PDM, VS1053
I (4832) main: Set baudrate at 115200
I (4832) main: Release 1.5, Revision 2
I (4832) main: SDK v3.3-dev-236-gfa59b1b1c-dirty
I (4832) main: Heap size: 152272
I (5172) main: Volume set to 35
I (5172) main: t0 task: 3ffcd814
I (5172) main: starting wifi
I (5182) wifi: wifi driver task: 3ffd71bc, prio:23, stack:3584, core=0
I (5182) wifi: wifi firmware version: abbd416
I (5182) wifi: config NVS flash: enabled
I (5192) wifi: config nano formating: disabled
I (5212) wifi: Init dynamic tx buffer num: 32
I (5212) wifi: Init data frame dynamic rx buffer num: 32
I (5212) wifi: Init management frame dynamic rx buffer num: 32
I (5222) wifi: Init static tx buffer num: 16
I (5222) wifi: Init static rx buffer size: 1600
I (5222) wifi: Init static rx buffer num: 10
I (5232) wifi: Init dynamic rx buffer num: 32
WIFI TRYING TO CONNECT TO SSID 2
E (5232) wifi: esp_wifi_disconnect 955 wifi not start
I (5262) main: connecting
I (5332) wifi: mode : sta (3c:71:bf:8c:01:78)
I (5332) main: Initialised wifi
I (5452) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:1
I (6462) wifi: state: init -> auth (b0)
I (6462) wifi: state: auth -> assoc (0)
I (6472) wifi: state: assoc -> run (10)
I (6482) wifi: connected with HomeLan, channel 11
I (6532) wifi: pm start, type: 1

E (6532) event: invalid static ip
E (6532) main:
Wifi connected

DNS: 192.168.178.1
IP: 192.168.178.58

I (7292) main: mDNS Init ok
I (7302) renderer: chip revision 1
I (7352) renderer: i2s intr:0
I (7862) main: RAM left 69560
I (7862) main: uartInterfaceTask task: 3ffe79f8
I (7862) main: clientTask task: 3ffe871c
I (7862) servers: watermark: 0xa4c  2636
I (7862) servers: telnetServer_sock socket: 54, errno: 0
I (7862) servers: Webserver socket: 55, errno: 0
I (7872) main: serversTask task: 3ffe9440
I (7872) main: task_addon task: 3ffeab48
Init ..I (8072) addon: event_ir: 3ffeae28
I (8072) addon: event_lcd: 3ffeacb0
I (8082) addon: rmt_nec_rx_task task: 0
I (8082) addon: task_lcd task: 3ffec128
........ Done
READY. Type help for a list of commands
autostart: playing:1, currentstation:5
##CLI.NAMESET#: 5 Radio Paloma
##CLI.URLSET#: www.radiopaloma.de
##CLI.PATHSET#: /listen.pls
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 35
I (10562) webserver: playstationInt: 5, device: 5
I (10862) webclient: Webclient socket: 57, errno: 0
##CLI.URLSET#: pool.radiopaloma.de
##CLI.PATHSET#: /RADIOPALOMA.mp3
##CLI.PORTSET#: 80
##CLI.STOPPED# from C_PLIST
I (12022) webclient: watermark : 3bc  956
I (12022) webclient: Webclient socket: 58, errno: 0
##CLI.ICY0#: Radio Paloma - 100% Deutscher Schlager!
##CLI.ICY3#: http:\/\/www.radiopaloma.de
##CLI.ICY4#: Schlager
##CLI.ICY5#: 128
##CLI.ICY6#: 100% Deutscher Schlager!
##CLI.PLAYING#
##CLI.META#: Nora Louisa - Nur Einmal
`
Next I updated to the latest version via telnet "sys.update". After that it failed again.
Then I went back to the V1.5R2 and it works again. So there is defenetly something different between these versions.
Do you have any further recommandations in order to track this issue down?

karawin, 2019-02-22:
The vlsi patch is the difference. 
Your vs crashs

Elbskipper, 2019-02-22:
OK. 
Which release is the latest one with the old patch?

karawin, 2019-02-22:
V1.6 R3

Elbskipper, 2019-02-22:
yep,
it works in V1.6R3 with the patch loaded. Interesting is, that the Buffer fill rate with the old patch is between 90 and 99% compared to 1 to 3% with the new patch.

karawin, 2019-02-22:
It is because the vs1053 crashed with dreq line activated so the buffer is read at the speed maximum. because the vs1053 seems to ask for data permanently.
If the vs1053 crash, the probable cause is a fake 1053b.
I don't think to go back to the old patch.
Sorry.

Elbskipper, 2019-02-22:
I didn't expect you to go back.
I will at a later time order another one. Although this one Looks exactly like the one you recommended.
Please consider this issue as closed.
[image]
Thank you!!!

karawin, 2019-02-22:
[image]

pilnikov, 2019-02-22:
red boards are better

karawin, 2019-02-22:
Alientek are better but production ended.

pilnikov, 2019-02-22:
я тут решил поставить жесткий эксперимент с обновлением со старой версии (1.3 наверное) сразу до актуальной 1.9R1 в качестве подопытного был достат с полки пылящийся без дела, подаренный китайцами  TTGO TS (не путать с TM) 
результат тут https://github.com/karawin/Ka-Radio32/issues/97 там же архив со схемой и конфигом
Кому лень переводить - работает. Но напрямую не обновилось, пришлось забивать сапогом (через прошивку бинарников)

Elbskipper, 2019-03-14:
Update:
I finally received the new board. This time I ordered the "red" one.
With this board it works fine. The Buffer fill rate is now at 92-98%.
I have no idea whats wrong with the "blue" board. I did check the DREQ connection on the board from the connector to the pin of the IC. It looks ok.
Regards

Hans
p.s.
the release V1.6 R4 worked as well with the "blue" board

closed 2019-04-09 by Elbskipper