karawin/Ka-Radio32 #98: Device leaks WiFi credentials in AP mode

r3d4, 2019-02-27:
On Ka-Radio32 1.9 when the configured Wifi is not available the device falls back to AP mode. In AP mode it is possible to connect with telnet to 192.168.4.1 and do a "wifi.station" command that shows the complete WiFi credentials. This is a big problem as AP mode is unencrypted. That means if the WiFi router is down, everyone in the neighborhood can readout my WiFi Key.

My suggestion is to completely disable telnet in AP mode.

karawin, 2019-02-27:
Good remark.
Thanks.
In the next release the wifi set will be disabled.
Thanks

r3d4, 2019-02-27:
Ok, Thanks, this sound good. As this is a real problem for me as Ka-Radio sometimes doesn't connect to my WiFi and goes to AP mode instead. Its automatically powered from my stereo, so sometimes it stays in AP mode for hours.

Btw, wouldn't it be better to disable telnet completely for security reasons? Maybe someone can abuse some buffer overflow...
Imho there is no point to use telnet when device is in unencryptet AP mode. In STA mode its quite handy.

karawin, 2019-02-27:
You can use the new command
wifi.auto
With this command, KaRadio32 stay on the Same AP and wait for the reconnection.

r3d4, 2019-02-28:
Unfortunately this doesn't help on boot. If there is no WiFi it goes to AP, independent of wifi.auto.
Meanwhile I  disabled telnet an build the sources, this seems to work for the moment.
Thanks

karawin, 2019-02-28:
Update again.
Done.

pilnikov, 2019-03-01:
Hello JP.
Again "thoughts" ;)
It may be possible to create a command for the serial port (wifi.autoap for example) to automatically switch to AP mode. And also run AP mode if the client's WiFi fields are not filled (empty). Or go to AP mode by pressing the "Prog" button (gpio 0) ???

r3d4, 2019-03-01:
> Update again.
Done.

Thanks!

@pilnikov 
I also like this idea to be able to prevent switching to AP mode. So the paranoid of us ( ;-)) can disable the AP mode with wifi.autoap ("0") or something like this.

pilnikov, 2019-03-01:
When and why do you need AP mode? 1. The first time to enter your WiFi data. 2. Not the first time - change your WiFi data. In other cases, you need Client mode. Right? Even if your home AP is down.

zuffnick, 2019-03-10:
could you put in a username/password for authentication of the telnet service?

pilnikov, 2019-03-10:
wifi.con("wifi_name","wifi_pass")

zuffnick, 2019-03-13:
Hi pinikov, 
Thanks for your comment. This was not what i meant.
Not every user of my network shall be able to read my wifi password. 

So I would prefer the following;

telnet $device
Welcome to Telnet-Service
Usename: <admin>
Password: ******

device>

it should be possible to set the password via cmd.

btw, IMHO the "Settings" tab in the web-if should also get an optional password protection

karawin, 2022-02-06:
Password are now not in clear text.

closed 2022-02-06 by karawin