karawin/Ka-Radio32 #102: Flashing on a new ESP32 results in "servers: Server low mem. Retrying..."

besi, 2019-03-08:
I am in the process of reviving an old radio:

[image]

In order to use the frequency wheel i need to modify Ka-Radio32. So the potentiometer will give me a voltage and based on that voltage I will switch the radio stations.

I have a running ESP32 which runs fine. I did flash the binaries onto it and it performs well. 

Now I have a new ESP32 from the exact same type and when I flash the binaries it boots up but then gives me endless errors when I access the web interface:

```
E (25316) servers: Server low mem. Retrying...
E (27316) servers: Server low mem. Retrying...
E (29316) servers: Server low mem. Retrying...
E (31316) servers: Server low mem. Retrying...
E (33316) servers: Server low mem. Retrying...
E (35316) servers: Server low mem. Retrying...
E (37316) servers: Server low mem. Retrying...
[...]
```

The same problem persists when I `make flash monitor` from source.

This problem keeps me from continuing which is a real shame. So any help tackling this would be greatly appreciated.

karawin, 2019-03-09:
A memory leak on your new code?
Hard to say something without information.
May it would be better to change the potentiometer to an encoder?

So I need the new code and all changed codes.
new gpio.h?
new hardware csv file?
etc.
To see a memory leak type
sys.heap time to time

besi, 2019-03-10:
The problem is I did not change the code but hardcode my network credentials so I won't have to connect to the created access point. 

Running `make menuconfig` did also change some config files, which you can all see here: https://github.com/besi/Ka-Radio32/commit/7064cd7f0f17ec9f6466257e3098965e2673e745

pilnikov, 2019-03-11:
The credentials that you have written in the sdkonfig file are not taken into account by the system.

karawin, 2019-04-10:
Closing, but fell free to reopen if you have some news.

closed 2019-04-10 by karawin