karawin/Ka-Radio32 #104: ST7567 SPI LCD display support

X-Stas-EEE, 2019-03-12:
Please add support for the ST7567 display to be available by telnet. 
For example: 'sys.lcd("74")'

The closest available option is ST7565, but it shows a mirrored and shifted picture.

Thanks!

karawin, 2019-03-13:
Update with serial or telnet
sys.prerelease

then test with one of these
#define LCD_I2C_SH1106		0 //128X64
#define LCD_I2C_SSD1306		1 //128X64
#define LCD_I2C_SSD1309		2 //128X64
#define LCD_I2C_SSD1325 	3 //128X64
#define LCD_I2C_SSD1306NN	4 //128X64
#define LCD_I2C_SSD1309NN	5 //128X64
#define LCD_I2C_SSD1306UN	6 //128x32
#define LCD_I2C_ST7567		7 //64x32

//SPI
#define LCD_SPI_SSD1306 		64 //128X32 (LCD_SPI =0x40)
#define LCD_SPI_SSD1309 		65 //128X64
#define LCD_SPI_ST7565_ZOLEN	66 //128X64
#define LCD_SPI_SSD1322_NHD		67 //256X64
#define LCD_SPI_IL3820_V2		68 //296X128
#define LCD_SPI_SSD1607			69 //200X200
#define LCD_SPI_LS013B7DH03		70 //128X128
#define LCD_SPI_SSD1306NN 		71 //128X64
#define LCD_SPI_SSD1309NN 		72 //128X64
#define LCD_SPI_ST7920 			73 //128X64
#define LCD_SPI_ST7567_pi 		74 //132X64
#define LCD_SPI_ST7567	 		75 //64X32

X-Stas-EEE, 2019-03-13:
Many thanks! Will try it as soon as I get home!

X-Stas-EEE, 2019-03-13:
I just have tried your solution. 
As I understand, my variant is the number 74. But right after choosing this option and reboot, the "KAR" logo was shown for a moment and then the whole screen was filled with black. 
I managed to notice that logo was shown correctly. Thus, I believe, we are on the right way 👍
It remains only to fix the black screen :)

karawin, 2019-03-13:
Do you have some other details for your board?
https://github.com/olikraus/u8g2/issues/440

X-Stas-EEE, 2019-03-13:
Well, a picture shown on the last display (called ZOLEN) looks similar to a picture I have now (mirrored and shifted). As I remember, I definitely have had a correct picture on my display using olikraus' library. I found  the demo sketch for Arduino IDE which make the display work correctly. It will be attached here.
The constructor line that can be useful is the following:
`U8G2_ST7565_NHD_C12864_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 18, /* data=*/ 23, /* cs=*/ 5, /* dc=*/ 2, /* reset=*/ 4);`
Please, don't pay attention to pin numbers.
Actually, I have two PDF files with some specs. Please, let me now if you need them to solve the problem.
[st7567.zip](https://github.com/karawin/Ka-Radio32/files/2963712/st7567.zip)

karawin, 2019-03-13:
U8G2_ST7565_NHD_C12864_F_4W_SW_SPI u8g2
Ok now i know what is the needed constructor.
May be tomorrow

X-Stas-EEE, 2019-03-13:
Thanks a lot!

karawin, 2019-03-13:
Done with 
#define LCD_SPI_ST7565_NHD_C12864	76 //64X32
sys.prerelease again.

X-Stas-EEE, 2019-03-13:
Now it works perfectly! Thank you very much!
Reviewed and CLOSED!

closed 2019-03-13 by X-Stas-EEE