karawin/Ka-Radio32 #105: [noob issue] Help interpreting GPIO Error

eqsone, 2019-03-22:
Hey there, thanks for sharing this awesome project! 

Though this was my first time leaving the Arduino environment, I was able to set this up within an hour, superb guidance. My initial setup was a Lolin32 lite, an SSD1306, DAC (PCM5102), and an Encoder (KY-040). This works like a charm. 

Now I tried to swap the DAC to a VS1053 (Adafruit), set the I2S GPIOS to 255 and enabled VSPI along with the VS1053 pin section to the recommended GPIOs. Now streams seem to play as well as before but there's no audio output, yet.

The serial output hints to an issue with my GPIO setup, I triple checked everything but can't seem to find the bug I obviously caused, some guidance in the right direction would be highly appreciated:

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3564
load:0x40078000,len:9372
ho 0 tail 12 room 4
load:0x40080400,len:5848
entry 0x400806d8
E (575) spiram: SPI RAM enabled but initialization failed. Bailing out.
I (138) main: starting app_main()
I (138) main: RAM left: 200080
E (138) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (158) main: Partition init done...
Log level is now ESP_LOG_ERROR
E (4398) gpio: GPIO_PIN mask error 
E (4398) gpio: gpio_set_level(158): GPIO output gpio_num error
E (4498) gpio: gpio_set_level(158): GPIO output gpio_num error
E (4598) gpio: gpio_set_level(158): GPIO output gpio_num error
WIFI TRYING TO CONNECT TO SSID 1
E (6328) wifi: esp_wifi_disconnect 964 wifi not start
E (7258) event: invalid static ip

DNS: 10.0.1.1  
IP: 10.0.1.18
(...)
`

My .csv file looks like this:

`gpio_space,namespace,,
K_SPI,data,u8,2
P_MISO,data,u8,19
P_MOSI,data,u8,23
P_CLK,data,u8,18
P_XCS,data,u8,32
P_RST,data,u8,12
P_XDCS,data,u8,33
P_DREQ,data,u8,34
P_ENC0_A,data,u8,255
P_ENC0_B,data,u8,255
P_ENC0_BTN,data,u8,255
P_ENC1_A,data,u8,14
P_ENC1_B,data,u8,15
P_ENC1_BTN,data,u8,13
P_BTN0_A,data,u8,255
P_BTN0_B,data,u8,255
P_BTN0_C,data,u8,255
P_BTN1_A,data,u8,255
P_BTN1_B,data,u8,255
P_BTN1_C,data,u8,255
P_I2C_SCL,data,u8,4
P_I2C_SDA,data,u8,5
P_I2C_RST,da ta,u8,255
P_LCD_CS,data,u8,255
P_LCD_A0,data,u8,255
P_LCD_RST,data,u8,255
P_IR_SIGNAL,data,u8,255
P_I2S_LRCK,data,u8,255
P_I2S_BCLK,data,u8,255
P_I2S_DATA,data,u8,255
P_TOUCH_CS,data,u8,255
P_JOY_0,data,u8,255
P_JOY_1,data,u8,255
P_LED_GPIO,data,u8,2
P_ADC_KBD,data,u8,255
P_BACKLIGHT,data,u8,255

option_space,namespace,,
O_LCD_TYPE,data,u8,0
O_LCD_ROTA,data,u8,0
O_LCD_OUT,data,u32,0
O_DDMM_FLAG,data,u8,1
`

pilnikov, 2019-03-22:
P_I2C_RST,da ta,u8,255

eqsone, 2019-03-22:
Holy crap, this was fast! Thank you so much, I'll try that.

eqsone, 2019-03-22:
That did it, stupid me. Thanks pilnikov, keep up with the good work!

closed 2019-03-22 by eqsone