karawin/Ka-Radio32 #106: [feature requests] Web-App capability | Reduced layout for small displays

eqsone, 2019-03-23:
Just my thoughts on potential minor tweaks, really appreciate the whole project as it is:

This within the html head should enable 'Karadio32' web-app capability with Apple mobile devices:

<meta name="apple-mobile-web-app-capable" content="yes">
<meta name="apple-mobile-web-app-status-bar-style" content="black-translucent">
<link rel="apple-touch-icon" type="image/png" href="logo.png">

The page could be saved to the homescreen within your given 'logo.png' App-icon and look/feel like a native App, without Safari's address and navigaton bars being shown. At least if set to a fixed IP.

Further I think readability of the 'now playing' screen with small displays like SSD1306 and SSH1106 might benefit from getting rid of additional metadata/date/time infos, reducing to the selected station, artist, song plus maybe volume and/or rssi levels. A sans-serif font might also help, though I have no idea if the related library is capable of doing so.

Awesome work anyway, thanks for sharing.

karawin, 2019-03-24:
Thanks for suggestions.
Not sure to understand the apple suggestion. 
I don't know the apple world.

eqsone, 2019-03-24:
I'm sorry, the html code in my comment wasn't shown without adding spaces after the left angle brackets.

Any browser but mobile Safari just ignores this on site load. Moblie Safari will show the site as fullscreen (web-) App instead of a standard browser site with navigation bars on top and bottom. But I must admit, it's quite impossible testing for this feature without Apple hardware unfortunately.

karawin, 2019-03-24:
Try the prerelease with the apple meta
```
<meta name="apple-mobile-web-app-capable" content="yes" >
<meta name="apple-mobile-web-app-status-bar-style" content="black-translucent" >
<link rel="apple-touch-icon" type="image/png" href="favicon.png" >
```
with the command:
sys.prerelease

eqsone, 2019-03-25:
Works. You're awesome.

- The used icon has a very low resulution, yet. I think about 120x120 should do it. Apple recommends much higher resolutions but 120 looks good enough in my humble opinion, at least on iP7/8 sized devices.

- "black-translucent" might be changed to "default". Translucent looks great until scrolling up, hitting white backgrounds. Default will show a white status bar on top with black phone-related text/symbols.

- Some minor issues left like some input tags supposed to be on the right side gets wrapped into the next line, therefore showing up on the left while held vertical. BUT this also appears within the regular browser view and I really think trying to sort all this out would take much more effort than it's worth. There aren't too many diy concerned Apple users out there who'd mind at all I guess. Most users will use Laptop/Desktop machines for their settings/administration anyway.

karawin, 2019-03-25:
try now.
I changed the logo to the 60x60 one,, and the default
Must be enough.

eqsone, 2019-03-25:
Beautiful, thanks for your effort!

closed 2019-03-25 by eqsone