karawin/Ka-Radio32 #110: Lilygo Epaper

nano111, 2019-04-12:
https://www.aliexpress.com/item/TTGO-ESP32-T5-V2-2-2-9-ePaper-Plus-Display-Module-e-Ink-Speakers-New/32850554753.html?spm=2114.10010108.1000014.1.140b65a7D7VXIN&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.13338.112237.000000000000000&scm_id=1007.13338.112237.000000000000000&scm-url=1007.13338.112237.000000000000000&pvid=ddf1094f-c593-4c02-b501-b2fb1ae7f2e8
Just bought one of this.No info fro thr display gpio ,all I found is 👎 

The connections to the eInk display do not seem to be listed anywhere, but I found that using an Arduino eInk library from https://github.com/ZinggJM/GxEPD , that the eInk worked without me having to change any of the pin designations in the example code.

Hence the pins used for the eInk display appear to be

CS/SS GPIO 5

DC GPIO17

RST GPIO16

BUSY GPIO 4

plus the default ESP32 SPI data and clock pins

Tried these gpios but not working.Any  help?
Also ,I flashed version 1.5 R1 and when I go to ota update it downloads and flashes but upoon reset I get:

I (38) main: Partition init done...
Log level is now ESP_LOG_ERROR
E (248) vs1053: VS1053 not used
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4009481a  PS      : 0x00060133  A0      : 0x80097905  A1      : 0x3ffb2d10  
A2      : 0xff050000  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060520  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0x3ffb2e00  
A10     : 0x00000001  A11     : 0x3ffc1f9c  A12     : 0x00000002  A13     : 0x3ffb42e0  
A14     : 0x00000000  A15     : 0x3ffb442c  SAR     : 0x0000000f  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xff050000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4009481a:0x3ffb2d10 0x40097902:0x3ffb2d40 0x400851c1:0x3ffb2d60 0x4008520f:0x3ffb2d80 0x40085e91:0x3ffb2da0 0x4000beaf:0x3ffb2dc0 0x40150f4b:0x3ffb2de0 0x400d8bda:0x3ffb2e00 0x400d2f60:0x3ffb2e40 0x40095aa5:0x3ffb2e60

Entering gdb stub now.
$T0b#e6

nano111, 2019-04-12:
https://github.com/Xinyuan-LilyGO/T5-Ink-Screen-Series
Maybe this helps.On version 1.5R5 all but display works.
Do you have to add support for this display?

karawin, 2019-04-12:
You need to configure the hardware for this board with a csv file
and choose the type 68 for the lcd
#define LCD_SPI_IL3820_V2		68 //296X128

nano111, 2019-04-12:
Thanks..
I was looking at a different gpio setup.Puting right gpio and lcd 68 did the trick.
Perfect.

closed 2019-04-12 by nano111

karawin, 2019-04-12:
Please send me the csv for other users.
Thanks
email:jp@karawin.fr

nano111, 2019-04-14:
Here is the csv for that board.
Btw..when I switch to clock the actual clock font is small.Can it be bigger?
Thanks

email:jp@karawin.fr<mailto:jp@karawin.fr>

karawin, 2019-04-14:
Thanks but where is it?
I don't have this display resolution, so i cannot test it.
Could you send a snap of the screen?
I can adjust the size per resolution in the code only.

nano111, 2019-04-14:
It is attached, isnt it?

Pozdrav
Goran

nano111, 2019-04-14:
[image1.jpeg]

[image2.jpeg]
Pozdrav
Goran

karawin, 2019-04-15:
Nothing received by github mail
Send to
mailto:jp@karawin.fr

Thanks

https://github.com/karawin/Ka-Radio32/issues/110#issuecomment-483062802