karawin/Ka-Radio32 #111: 32bit I2S DACs not handled correctly

har-in-air, 2019-04-19:
In audio_renderer.c : render_samples() function, for a 32bit output stream, outBuf8 needs to be malloc'ed with double the size of the input buffer, not the same size, as for every 16bit sample, there is a 16bit zero padding.

As the code stands, it will cause a memory fault on selecting I2S_MERIUS (32bits) when playing a stream.

I've fixed this and its working fine now with my homebrew I2S power amplifier using TAS5753MD chip (which requires a 32bit input stream for other reasons).

++
	uint32_t outBufBytes = buf_len*(2/buf_desc->num_channels);
       if (renderer_instance->bit_depth == I2S_BITS_PER_SAMPLE_32BIT) outBufBytes <<= 1;
	outBuf8 = malloc(outBufBytes);
++
and at the end of the function you need the same  fix 
++++
	if (bytes_written != outBufBytes) ESP_LOGV(TAG, "written: %d, len: %d",bytes_written,bytes_left);
        bytes_left -= bytes_written;
++++

karawin, 2019-04-19:
Ok, thanks

closed 2019-04-19 by har-in-air

karawin, 2019-04-19:
if (bytes_written != bytes_left) ESP_LOGV(TAG, "written: %d, len: %d",bytes_written,bytes_left);